makerbase-mks/MKS-TFT #313: MKS TFT 70

RobertoBrachini, 2019-04-29:
Is it possible to have the TFT70 BMPs? or the MakerBase Image Editor updated to open BINs. Thank you

tomaran, 2019-05-04:
Very frustrating there is practically zero support for the TFT7.0 - for the record if it helps you.
the bmp_logo is 800x480 dpi and the rest of the icons are 100x100 dpi. If you change the background colour in the mks_config.txt file to something other than the default, you will notice that they are actually smaller than what the icon would suggest (basically the background button is LARGER than the actual icon itself). With this in mind, I took the TFT3.5 icon bmps' and created my own in photoshop. I then used Image2LCD to confirm the bmp to .bin files. 100 x 100 dpi 16-bit colour. I'm happy to upload some pictures of the UI and if interested I will upload a zipped copy of my mks_pic folder. Let me know. Either way the default buttons are depressing to look at ha!

RobertoBrachini, 2019-05-05:
thank you,
I was interested in the original bmp to be able to customize them

tomaran, 2019-05-08:
I see, as you know the image editor does not work with the tft3.5 or tft7.0 bin files. This is why I took the actual 3.5 bitmaps and manipulated them myself. The tf7.0 are pretty bland anyway so not sure why you would want to customize. Hopefully they will update the app soon!

RobertoBrachini, 2019-05-09:
I wanted to customize the "more interface" menu and make icons similar to the original ones

kleverline, 2019-05-09:
tomaran, I am having issues with changing the boot screen on my TFT35.  The image looks fine on my PC, but is all sorts of garbled when displayed on the TFT35 itself.  Is this the issue you were referencing when you said "I see, as you know the image editor does not work with the tft3.5 or tft7.0 bin files"?

rafaelferrari0, 2019-05-15:
Hi, I use BinImageViewer.exe to view TFT70 BIN files (100x100 px), then printcreen and photoshop.
Finally Img2Lcd.exe convert BMP to  BIN with these parameters:
- Binary
- Horizontal scan
- 16- bit truecolor (below 4096)
- 100 x 100
- all checkboxes off

For Boot LOGO, change size to 800x480.

RobertoBrachini, 2019-05-16:
hi and thank you so much, after I try

RobertoBrachini, 2019-05-17:
perfect!

RobertoBrachini, 2019-05-20:
do you know the "bmp _ ??? _ state" icon sizes?

RobertoBrachini, 2019-05-20:
possono essere usati altri colori esadecimali oltre a quelli in elenco? (16)

RobertoBrachini, 2019-05-20:
can other hexadecimal colors be used besides those in the list? (16)

Ghenghis, 2019-07-26:
> Very frustrating there is practically zero support for the TFT7.0 - for the record if it helps you.
> the bmp_logo is 800x480 dpi and the rest of the icons are 100x100 dpi. If you change the background colour in the mks_config.txt file to something other than the default, you will notice that they are actually smaller than what the icon would suggest (basically the background button is LARGER than the actual icon itself). With this in mind, I took the TFT3.5 icon bmps' and created my own in photoshop. I then used Image2LCD to convert the .bmp to .bin files. 100 x 100 dpi 16-bit colour. I'm happy to upload some pictures of the UI and if interested I will upload a zipped copy of my mks_pic folder. Let me know. Either way the default buttons are depressing to look at so I had to do something about that ha!

I would be interested in your updated BMP folder you created, I just ordered one of these.

SebastianCardenasEscoto, 2021-02-20:
> tomaran, I am having issues with changing the boot screen on my TFT35. The image looks fine on my PC, but is all sorts of garbled when displayed on the TFT35 itself. Is this the issue you were [...]

I'm having this same issue. Did you find a fix?