makerbase-mks/MKS-TFT #318: MKS_TFT_CNC can't move Stepper motor

chenmaker, 2019-05-16:
I have a TFT2.4 and MKS DLC v2.0
The  MKS_TFT_CNC burn in mks dlc 
But i tounch TFT2.4 move  the stepper motor can't move

closed 2019-05-28 by chenmaker

selimayaz, 2020-01-13:
do you have any soluiton?

makerbase-mks, 2020-01-14:
Can you use the pc software GRBL to control the movement of the DLC motor normally? If possible, please confirm whether the baud rate of TFT24 and DLC motherboard are consistent.

selimayaz, 2020-01-14:
Yes, i can use with pc cncjs. Baud is 115200 both. Screen says connected. But nothing?

chenmaker, 2020-01-17:
Because TFT need a program and DLC motherboard need too
TFT-->https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS_TFT_CNC
DLC motherboard-->https://github.com/grbl/grbl

DukeQuincy, 2025-03-27:
I had the same problem. By default the carving speed is set to 0%. You can change this by clicking on the second icon on the bottom row and then using the + button to set the speed to 100%