makerbase-mks/MKS-TFT #320: MKS TFT32 and smoothieware ... temperatures troubles

hatoupix, 2019-05-21:
Hi all, 

I've put the last firmware on my TFT32 (v3.0.3) and my smmothieware is 4th December 2018
(communication rate are the same 115000)
I can move axes with tft ...

But my problem is that temperatures are not taken into account (manually and by gcode). 

If I start a print, the temperature increase BUT not displayed the REAL one (it still 21° (example)) (same problem with manual increase temp)
AFTER if I make a reset, the temperature displayed is by example 100° (cause just before I send a begining of print ...) ... but not decrease

same pb with bed temperature 
And on my LCD 12864 all temperatures are fine and increase/decrease perfectly !

Thanks for your helping

B3N100, 2019-05-22:
https://github.com/makerbase-mks/MKS-TFT/issues/321
Have we got the issue ?

hatoupix, 2019-05-22:
not yet ...

GageHughes, 2019-06-03:
Im having the same problem where the tft 32 stops sending code to the board after I touch print all the temperatures on the tft stay the same and I cant adjust anything. The temperature of the printer does increase though.
My printer runs marlin 1.1.x bugfix on a MKS GEN 1.4 and the tft has v3.0.3 firmware.