makerbase-mks/MKS-TFT #321: TFT32 Refresh SGEN, Marlin 2.0

B3N100, 2019-05-21:
Hi,
I use Sgen motherboard with TFT32 and Marlin 2.0, all work without printing from SD card/usb slot and refresh of the temperature.

In fact, when i heat, hardware heat, but screen not, like no refresh a page. When i change function, and come back, temperature is udapte, but if i do anything, display temperature no change.
Due to this proprem, i can't print from any device other than my computer.

Thanks to your help, i would like to fix this annoying issue.

Sorry for my english i'm french ^^'

tduehr, 2019-06-28:
I have the same problem with a MKS Gen L and Marlin 2.0.x in a BIBO Touch X. It seems line the TFT is losing sync with the board. I couple of commands will go though then it stops sending more. I know it's parsing at least the first few lines of gcode because the temps from the file show as the target temps but the actual temperature either shows as 0 or room temp. If I preheat the extruders through the TFT32, then print, the hotends will get to temperature but nothing else happens.

NeoMod, 2020-06-25:
I can confirm: SGEN_L with Marlin 2.0.x and TFT2.8.
Definitely seems a sync problem with the motherboard.