makerbase-mks/MKS-TFT #324: TFT35 : color of temperature,fan,(etc) background not working

francoisbrassart, 2019-06-04:
Hi,
I'm trying to change color of temperature,fan,(etc) background on my TFT35 but it's not working, this background stays black.
When I change the color of other buttons, background... it works well.

For example, this is a config file I used to try to change the color of temperature,fan,(etc) background in red : 
```
#--------------------------------------------------------------------
#########  Printer type setting  ##################################

#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
>cfg_firmware_type:1

#machine setting (Normal:1; Delta:2)
>cfg_machine_type:1

#baud rate (9600:1; 57600:2; 115200:3; 250000:4)
>cfg_baud_rate:4

#multi-language(enable:1, disable:0)
>cfg_multiple_language:1

#languages setting(simplified (simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5;French:6;Italian:7).
#This configuration is valid when "cfg_multiple_language" is disabled.
>cfg_language_type:3

#extruder number(one:1; dual:2)
>cfg_sprayer_counter:1

#enable heated bed(yes:1; no: 0)
>cfg_custom_bed_flag:1

#the max target temp of extruder and heated bed
>cfg_max_sprayer_temperature:270
>cfg_max_hotbed_temperature:150

#pause position (-1 is invalid; Z-axis is relative position)
>cfg_XPOS:-1
>cfg_YPOS:-1
>cfg_ZADD:10

#-----------------------------------------------------------------------
#########  Advanced Function Option  ##################################

#enable UPS?(yes:1; no:0)
>cfg_have_ups_device:0

#enable power detecting module(mks 220det:1; mks pwc:0)
>cfg_insert_det_module:0

#enable auto off after print finish function (no:0; Yes:1）
>cfg_print_finish_close_Machine:0

#set PB0 signal when use it for sencond nozzle (high level:1; low level:0)
>cfg_PB0_trigger_Level:0

#set PB1 signal (high level:1; low level:0)
>cfg_PB1_trigger_Level:0

#set error range of Z-axis on breakpoints recovery 
>cfg_breakpoint_z_error:0.2

#the function of power detecting and filament detecting(1:disable;0:enable)
>cfg_mask_PB0_PB1_function:0

# the method of save the gcode data with power off(1:Capacitor storage; 0:File system save)
>cfg_pwroff_save_mode:0

#---------------------------------------------------------------------------

#########  Filament Change Function  ##############################

#the speed to extrude filament(mm/min)
>cfg_filament_load_speed:1200
#the lenght to extrude filament (mm)
>cfg_filament_load_length:200

#the speed to retract filament(mm/min)
>cfg_filament_unload_speed:1200
#the lenght to retract filament(mm)
>cfg_filament_unload_length:200

#It is the minimum temperature for filament change.
It will auto heat up if the current temp doesn't reach the target.
>cfg_filament_load_limit_temperature:200

#---------------------------------------------------------------------------

############  Leveling Function  ########################################

#leveling mode(manual:0; auto:1; conceal leveling button:2)
>cfg_leveling_mode:0

#the command of auto leveling (G29 is available for Marlin.While G32 is for  Repetier and Smoothieware)
>cfg_auto_leveling_cmd:G28;G29;

#the point number of manual leveling:(3,4,5 point available)
>cfg_point_number:4

#the coordinates of 5 point on manual leveling
>cfg_point1:50,50
>cfg_point2:180,50
>cfg_point3:180,180
>cfg_point4:50,180
>cfg_point5:150,150

#the travel speed of leveling(mm/min)
>cfg_leveling_z_speed:1500
>cfg_leveling_xy_speed:3000

#When the XY axis is moved manually, the height of the z-axis of the extruded Z axis from the origin (mm)

#normally only needs to be reconfigured when the z axis is zero to the maximum under the smoothware firmware.
> cfg_leveling_z_high: 10

#---------------------------------------------------------------------------
 
#############  WIFI Function  ##########################################

#wifi mode(AP:1; STA:0)
>cfg_wifi_mode:1

#wifi name and password
>cfg_wifi_ap_name:MKS_wifi
>cfg_wifi_key_code:makerbase

#0:disable cloud service 1:enable cloud service
>cfg_cloud_enable:1
#cloud service adress
>cfg_wifi_cloud_host:www.baizhongyun.cn
#cloud service port
>cfg_cloud_port:10086

#Dynamic access IP
>cfg_ip_dhcp_flag:1

#--------------------------------------------------------------------------------

############## Function Customization #######################################
#functional button:
#in the style of simple, the functional button display in the "set" interface
#in the style of Classic,the functional button display in the "home" interface
#functional button(0:no display; 1:display)
>cfg_function_btn1_display:1

#the functional button 1 command
>function_btn1_cmd:M84;

#The "stop" button below the "Home" interface(0:disable,1:enable)
>cfg_quickstop_btn_display:0

#The "zoffset" button below the "More" interface(0:disable,1:enable)
>cfg_zprob_offset_btn_display:0

#set number of  "More" button 
>moreitem_pic_cnt:0

#edit command for 1~7 "More" button,each command must be separated by semicolon ";"
>moreitem_button1_cmd:G28 X0;

>moreitem_button2_cmd:G28 X0;

>moreitem_button3_cmd:G28 Y0;

>moreitem_button4_cmd:G28 Y0;

>moreitem_button5_cmd:G28 Z0;

>moreitem_button6_cmd:G28 Z0;

>moreitem_button7_cmd:G28;

#how many "More" button display on screen interface
>morefunc_cnt:0

#command setting on button1~7
>morefunc1_cmd:G28;
>morefunc2_cmd:G28;
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;

>morefunc7_cmd:G28;
#---------------------------------------------------------------------------------
##### Display Customization(button and logo customized see User's Manual) #######

#the color of screen background
>cfg_background_color:0x000000

#the color of title text
>cfg_title_color:0xFFFFFF

#the color of temperature,fan,(etc) background
>cfg_state_bkcolor:0x0000FF

#the color of temprature,fan,(etc) text
>cfg_state_text_color:0xFFFFFF

#the color of file button
>cfg_filename_bkcolor:0x000000

#the color of file text
>cfg_filename_textcolor:0xFFFFFF

#the color of general button background
>cfg_btn_bkcolor:0x000000

#the color of general button text
>cfg_btn_textcolor:0xFFFFFF

#the color of status button background
>cfg_state_btn_bkcolor:0x000000

#the color of status button text
>cfg_state_btn_textcolor:0xFFFFFF

#the color of "back" button background
>cfg_back_btn_bkcolor:0x000000

#the color of "back" button text
>cfg_back_btn_textcolor:0xFFFFFF

#the color of selected button background
>cfg_sel_btn_bkcolor:0x000000

#the color of selected button text
>cfg_sel_btn_textcolor:0xFFFFFF

#the color of dialog button background
>cfg_dialog_btn_bkcolor:0xff0000

#the color of dialog button text
>cfg_dialog_btn_textcolor:0xFFFFFF

#3D effects setting (disable: 0; enable: 1)
>cfg_BUTTON_3D_effects:0

#position of button font offset bottom(unit:pixel)
>cfg_btn_text_offset:23

##Screen rotation 180 °(0:disable,0xEE:enable)
>cfg_screen_overturn_180:0x00

#the style of display on the screen home(0:classic,1:simple)
>cfg_screen_display_mode:1

#Enable function of display error message. (no:0; Yes:1）
>cfg_display_eeror_message:0

#-----------------------------------------------------------------------
```

kleverline, 2019-10-18:
I can confirm this issue.  I can also tell you that I have asked about color issues/config setting not working and MKS has not offered any fixes (nor have they provided updates or fixes for the litany of other problems with this display).  This product is essentially "abandon-ware".

https://github.com/makerbase-mks/MKS-TFT/issues/305

makerbase-mks, 2019-10-22:
You can try to format the SD card, then copy the configuration file to the SD card again, power on or reset the motherboard.

MarshMalloW0007, 2019-12-22:
Hello,
Read my issue to solve your problem.
Marsh