makerbase-mks/MKS-TFT #325: TFT24: Extract - Reload SD Card

scmanjarrez, 2019-06-07:
Hi,
Would be possible to implement an option to easily remove sd card and then reload the screen to load new files? Having to restart the printer in order to extract/load sdcard new files isn't comfortable.
Thanks.

P.S: Any plan to release source code so we can implement our own features? That would be great to make the tft community bigger :smile:

MKS-Sean, 2019-11-13:
Get the gcode file does not need to be restarted

scmanjarrez, 2019-11-13:
Sometimes when you extract the sd card, upload the gcode and connect again the sd card, the screen goes black having to restart the printer.

makerbase-mks, 2020-03-12:
We are preparing materials for open source code, please wait for our good news!

closed 2020-03-12 by makerbase-mks