makerbase-mks/MKS-TFT #331: MKS GEN 1.4 + TFT32 wifi shield + BLTOUCH  > USB ok ? (for octoprint)

zagg, 2019-06-21:
Hello,
I have actually MKS GEN L + TFT32 + MKS WIFI + BLTOUCH , I can't use TFT32 when I want use Octoprint via USB. I must disconnect the TFT 

Can I use MKS WIFI or other way to connect TFT32 + USB simultaneous ?

If I buy MKS GEN 1.4 , same problem exist ? TFT32 and USB use same port on this board ?

Thanks for help.
Regards,
Davy

jesusperiago, 2019-07-28:
Did you try to uncheck
 ```
Request exclusive access to the serial port
Uncheck this if you are having problems connecting to your printer.
Serial logging
```

makerbase-mks, 2020-03-12:
The USB and MKS TFT AUX-1 interfaces are shared, so these two interfaces cannot be used at the same time. Sorry!

closed 2020-03-12 by makerbase-mks