makerbase-mks/MKS-TFT #337: MKS-TFT35 V1.0.2  still and alway print random stop

wiwatto, 2019-08-01:
MKS-TFT35 V1.0.2  firmware issues ?

My system : MKS SGen + Smoothieware + MKS-TFT35 V1.0.2 

MKS-TFT35 V1.0.2  STILL and alway stop random 
 it work good printing for a long hours  no problem with LCD-12864

kleverline, 2019-10-18:
MKS has abandoned the TFT35.  Sadly, you should not expect any help or fixes from them.  I would recommend you do what I did.  Go buy a BigTree lcd for your printer and then use darkspr1te's alternative firmware to re-purpose the TFT35 for some other use (or toss it in the trash and consider it a lesson learned about how MKS treats their customers).

https://github.com/darkspr1te/MKS-TFT-V4-alt-fw

wiwatto, 2019-10-30:
kleverline thank for information. i agree with you about MKS treats their customers

evandene, 2020-02-24:
Yep toss the screen and install the good old 12864 screen.
Over there they have no clue about customer service. They simple show disrespect for any customer.  
For to make the 12864 to work on your MKS Sbase v1.3 just cut away a notch out the EXT1 male connector FROM THE 12864 screen in order to make the pin adjustments for the EXT1 cable.

makerbase-mks, 2020-02-25:
Hello, may I ask what version of firmware you use for Smoothieware ？