makerbase-mks/MKS-TFT #338: No communication between TFT28 and MKS Gen L

mremingtonb, 2019-08-05:
Hello.  My TFT28 has stopped communicating with my MKS Gen L.  It used to work fine, but I had to rewire it.  Since putting it back together, the TFT28 won't communicate with the MKS Gen L anymore.  I'm fairly certain that I have my wiring correct.  The baudrate on both the TFT28 and MKS Gen L is set to 115200.  Any ideas what is going on?

Stady234, 2019-12-10:
Not sure if I can help any.. but what 'rewire" did you do?

mremingtonb, 2019-12-10:
> Not sure if I can help any.. but what 'rewire" did you do?

No need to worry.  I eventually got it to work by changing the baudrates over and over again on the board and screen until they finally just clicked.  I think 250000 finally worked if I remember correctly.  It's weird, I changed between that and 115200 several times on both the board and screen, and then one time they both just decided that 250000 was right.

closed 2019-12-10 by mremingtonb