makerbase-mks/MKS-TFT #342: Open the source, please!

javikas, 2019-09-03:
It's been asked a few times, but I have to insist again: When are you going to open the source and let us improve this firmware?
Do you realize that many of us are experienced programmers, members of the RepRap community, makers of our own printers and firmwares, and many of us find this firmware ugly and unprofessional?

You already need open source projects to sell boards, without Marlin (among others) you won't be there so, why do you think this way for the MKS-TFT?

This is all about RepRaps, we want to do things our way. We don't want this "four screens with buttons and ugly fonts thing" any more.
I've even seen you answering to remove the buzzer from the board as a solution to cancel the starting beep. Really?

Come on friends, stop and rethink your business model. We can help you. Many other printer brands will look at you if they see a professional looking screen instead of... this!

makerbase-mks, 2019-09-03:
Sorry for that!
Due to the copyright of some code in the tft series, we can not directly open source related code.

javikas, 2019-09-03:
Even if part of the code is copyright protected, you could do some kind of mix between open and closed source. Look at the Chromium project, even Google integrates open and closed source developing Chrome.

Are you talking about core libraries (low level controllers) or high level libraries of any kind?

Our main interest would be to have the basic structure of controllers and connections, not the hierarchy of menus or any library involved in it.

If you maintain a second git with this core structure, you could benefit from it. Not only because more people would move to your hardware but also because you can implement all kind of stuff to your closed one, and for sure you'll end up moving your copyright'ed code to open source developments.

I don't plan to bother you again with this issue, and I thank you for your response. My only goal is to show you that there's another business model increasingly adopted by the biggest corporations. This is not a "concession" that you can grant us but the smartest and cheapest way for a company to grow up and earn the love and loyalty of a whole community. Please think about it.

UnderXirox, 2019-09-04:
bonjour
il y a du monde qui demande que ca l'ouverture du code
pour
remarque : 
comme la marque BigTreeTech en autre qui fait la même chose que vous mais lui il et open source
donc résultat Les personne vont se tourne ver de l'open source normale

donc le résultat final pour vous sera la fin de vente pour vos Ecran

InsanityAutomation, 2019-09-17:
When the question on host prompt support to allow filament runout and such to function went unanswered, I just moved to the BigTreeTech displays anyway, as theyre not only dual mode but all source is open. Perhaps it can be ported to this hardware, albeit without the 12864 emulation mode as they dont have an encoder or button.

p1ngb4ck, 2019-10-04:
forget about it. makerbase could open sources if they wanted to - th problem is just completely false (dumb) thinking in their management.
They are still having the 1960s view on things, and think they could easily sell products by copying open-source pojects, closing them by doing useless and errorprone modifications and then sell & protect their "intellectual" property. I am finished with makerbase. Selling unfinished unuseable products and then take away the option to at least make the hardware work with s.o. own firmware - because all necessary info is kept secret - this just sucks. in an epic way

kleverline, 2019-10-18:
Yeah, I will never buy an MKS product again after my experience with the abortion of a product that is the TFT35.  It literally just flat doesn't work.  The firmware is trash, essentially none of the settings in the config file actually do anything, and it won't communicate properly at all if anything else is hooked to the motherboard that uses serial, even if the other device isn't actively sending data.  The biggest problem though is that MKS has essentially completely abandoned the device.  No updates, no (useful) help on bugs/complaints, etc.  

One more time, makerbase-mks, **I will never buy another product from your company and will actively discourage others from buying them as well.**  There is WAY too much competition in this industry for you to be able to justify the half-finished/half-working products and total lack of support.

quarkpower, 2019-10-19:
https://www.youtube.com/watch?v=GIU4yJn2-2A

Underp4ntz, 2019-11-07:
Just buy the SKR Screens from BIQU / BigTreeTech. you can edit i guess everything since you have to compile your own firmware and other stuff. When i noticed you can only do basic settings with the MKS Screens i was like... Ah well... What a waste.

Again do not buy MKS Screens if you want to have "full control" and compile your own firmware.
And a copyright law in China? Really?

I tried to flash the firmware it was all good but now my screen reacts really strange when pressing buttons... like there is a windscreen wiper moving over it...

stef-ladefense, 2019-11-28:
let them die

darkspr1te, 2020-01-01:
For those wishing to develop for the MKS TFT series devices , here you go 
https://github.com/darkspr1te/MKSTFT_Marlin_Touch
Firmware is almost 100% complete, just a few issues to fix
Forum for discussions 
https://mcu.selfip.com/viewtopic.php?f=6&t=202&p=1643#p1643
darkspr1te

darkspr1te, 2020-01-01:
> Sorry for that!
> Due to the copyright of some code in the tft series, we can not directly open source related code.

We Have solved this problem for you and provided a open source touch firmware for Marlin + others. 
We have also created board bring ups environments in Arduino/PlatformIO with all the libraries to use the on-board hardware like touch (with calibration) USB Host, uarts, SPI, eeprom, flash, bootloader, LCD and sound. 
For fun there is also a port of flappy birds and tetris. 
For hardcore setup we have Wifi flashing with esp2866/esp32, wifi uarts + more 
DMA+watchdog and RTC functions 
proper segfault handling with backtrace 
code works with or without factory bootloader, if you use factory bootloader then you can start developing right away without additional hardware , although a stlink/jtag adapter is recommended.  
if you wish us to port to the other LCD screens then please provide us with sample hardware to develop on.
[most frustrating game in the world](https://github.com/darkspr1te/vscode_arduino_mkstft_flappyduino.git)
[Basic board profile platformio](https://github.com/delwinbest/MKS-TFT28
)

darkspr1te 
delwinbest

darkspr1te, 2020-01-18:
Good news everybody , 
I have just finished working on open source bootloader , see my repo for both a opensource bootloader and open source firmware :-)
https://github.com/darkspr1te/stm32f107vc_sd_bootloader
default settings for mkstft28 v4 board 
darksrp1te

scmanjarrez, 2020-03-24:
Guys, makerbase has released the source code, check the readme.

Failx92, 2020-09-15:
Hello, how can I compile a .bin with the source code? I would like to add German language for my tft28 display. 
Regards

darkspr1te, 2020-09-15:
To compile MKS TFT firmware int to a .bin  you need to download IAR Embeded workshop  https://www.iar.com/iar-embedded-workbench/, or there is alternative firmware with german already added at bigtreetech repo and the MKTFT32" is supported

Stellarspace, 2020-10-28:
> Guys, makerbase has released the source code, check the readme.

Thank god!