makerbase-mks/MKS-TFT #352: MKS TFT32_L V3.0

Allimm, 2019-11-01:
After turning on the screen, booting ... appears, after which the screen turns black.
What could happen?
Re-flashing does not help.

makerbase-024, 2019-11-05:
If the firmware is re-updated, the same situation will occur, which may be a hardware problem.

Allimm, 2019-11-05:
> If the firmware is re-updated, the same situation will occur, which may be a hardware problem.

which part on board i should to replace?

Allimm, 2019-11-06:
?????????????????????

makerbase-024, 2019-11-07:
The problem may be the chip in the D10 position.
[image]

Allimm, 2019-11-07:
Thanks!
 If the matter in flash memory is out of order, where can I download the contents of the memory to flash the chip.

makerbase-024, 2019-11-08:
Just need to re-flash the firmware.(mks_pic,mks_font,mks_tft28.bin,mks_config.txt).

Stady234, 2019-12-10:
did you get your issues solved?