makerbase-mks/MKS-TFT #353: MKS tft32 M600 support

docean24, 2019-11-13:
Can support please be added for M600 commands? Currently M600 will pause a print but the touch screen does not recognize that the print has paused and there is no way to resume the print. This is for advanced filament out functionality so assigning a "more" button will not help.

makerbase-mks, 2019-11-15:
This feature, we can implement in the next release. 
This logic is M600-->display pause and machine pause---->click on the touch screen to resume printing. Do you need such a function?

marcohaddad, 2019-11-15:
Hello,
That's great! I'll love to have it too!
Best regards,

docean24, 2019-11-16:
Yes that's what I need thank you. When will the next release be?

Stady234, 2019-12-10:
I have been waiting for almost a year for this to be added... more specifically... if ANY printer pause/user must click old screen button to continue - option to be added.. but still waiting...

makerbase-mks, 2020-01-13:
We are preparing for open source work in 2020. I think if we have time, we will definitely do this.

manukieli, 2020-01-16:
Very good new.
M600 command will helpfull.

rost76, 2020-01-17:
I am also waiting for tft32

Claymore61, 2020-01-20:
I would lime this function too.
I was thinking I just hadn’t coded it correctly 😂😂

Claymore61, 2020-01-20:
It of interest, when is the next release likely to be ?
I’m very new to this!

bartofborg, 2020-02-07:
this is needed! any time line revision yet?

tanaes, 2020-02-11:
Adding my interest for this feature as well!

Telekmac, 2020-04-17:
Please add this. You should add support for all functions that require user input sent from Marlin. Especially m600

betocabus, 2020-05-21:
M600 is the only funtion I want to change from traditional LCD to touch LCD, all of them doesnt work! I use M600 every day!