makerbase-mks/MKS-TFT #354: MKS TFT 70 V1.1 not working with SKR 1.3

Mercer-PL, 2019-11-19:
Printer: Ender 3 (sort of, after heavy modifications)
Motherboard: SKR 1.3
Firmware: Marlin bugfix 2.0

Screen firmware: 1.0.1 and 1.0.2
Screen power supply: 12V Redox power supply

Issue:
1. The board and screen are not communicating with each other. I''ve updated LCD settings to match baudrate of my motherboard (115200). Ribbon cable is connected to AUX-1 on SKR 1.3 and on screen. But, the screen displays the temperature set for hotend as 0/180 and increasing the set temp doesnt't chane anything (I monitor my settings on Full Graphic smart controller and the temperatures are set to 0). What's more, the TFT screen doesn't read any temperatures from printer, displays 0 degrees instead of 19.

[mks_config.txt.txt](https://github.com/makerbase-mks/MKS-TFT/files/3865212/mks_config.txt.txt)

2. I've changed my levelling settings to auto (with G28 and G29 for Marlin). But after the update in levelling menu I have buttons for positions 1, 2, 3, 4. 

3. When I've changed the language setting to single language and english - the language menu is still showing all the languages and eanglish as set. Should it be like that?

makerbase-mks, 2019-11-25:
You need to make sure your profile name is "mks_config.txt".it does not the"mks_config.txt.txt"

Mercer-PL, 2019-11-25:
Yes, that was the issue. I needed some time, it was so stupidly obvious i haven't thought about it. Sorry for the mess.

closed 2019-11-25 by Mercer-PL