makerbase-mks/MKS-TFT #355: No booting

Tuxomx, 2019-11-21:
Hi. I have a TFT28 V2.0 with firmware updated some time ago, it was working flawlesly. Today I tried to use it and it only beeps and shows a black screen. I have been unable to update firmware. What can I do to fix this?

Stady234, 2019-12-10:
have you tried no sdcard in screen.. and only the aux cable connected to board?  What does it do when you try to re-flash the firmware?

darkspr1te, 2020-01-01:
> Hi. I have a TFT28 V2.0 with firmware updated some time ago, it was working flawlesly. Today I tried to use it and it only beeps and shows a black screen. I have been unable to update firmware. What can I do to fix this?

You can reflash bootloader and firmware using simple uart tool(ramps board,arduino,buspirate) and guides from my github, https://github.com/darkspr1te

makerbase-mks, 2020-04-01:
Hi! Has this issue been resolved?

closed 2020-05-27 by makerbase-mks