makerbase-mks/MKS-TFT #356: MKS TFT35 problems after update

DELAS-bit, 2019-11-22:
I have update my mks tft35 with github file.
After update lcd don't show the icon.
Someone can help me?
[image]

Stady234, 2019-12-10:
while this is not specific to your screen size.. maybe it will help.. I used it to configure mine and had no issues...  https://youtu.be/ncc7TRkCH-M

aokiakagome, 2019-12-15:
[image]
[image]

VegetableKingdom, 2020-01-09:
I think I know what caused your situation.
You have downloaded and uncompressed the firmware using an Imarc computer.
Because Apple Computer will generate an extra backup file when decompressing the file. When the screen updates the picture, the number of files will double. So update error.
As long as you switch to the windows system to download and decompress the new firmware. Then update tft35 again. This situation will be resolved