makerbase-mks/MKS-TFT #357: MKS TFT35 Problems

JanHH2019, 2019-12-06:
I tried to change the icons in the TFT35. Unfortunately, this did not work and now after loading the icons from the directoryMKS-TFT-master \ MKS-TFT3.5 \ TFT35_V1.0.2 Release file \ TFT35_V1.0.2 Release file \ Images \ mks_pic the icon for Autolevel is highlighted in blue and works no more.
Where can I get the original icons?
what kind of pixel format do the icons have to have?
the icons of the other versions do not fit either.

JanHH2019, 2019-12-06:
After a long search, I have now found the original version robin2 TFT35 and recorded. everything is running again.
but how do I create a new icon?
the software MakerBaseImageEditor generates from the * .PNG 16 KB large BIN files. the original ones are 32 kb, the software Img2Lcd
generates c - source code.
I can not do anything with both.
So where can I find an explanation and the software to get the custom icon's on my own?

Stady234, 2019-12-10:
Try this... I followed this for my TFT32... should be same... https://youtu.be/ncc7TRkCH-M?t=371

aokiakagome, 2019-12-15:
[image]
[image]

Original icon↓
https://github.com/makerbase-mks/MKS-TFT/blob/master/UI-Design/TFT35_BMP_FILE.rar

GunnarFrei, 2020-01-12:
I just managed to get it working on my TFT35 like this:

- Save your logo as 480x320, bpm (24bit)
- convert in Img2Lcd with the following options:
`Binary
Horizon Scan
16-bit TrueColor (first one in drop down)
480x320
Include head data`
save as bmp_logo.bin in the correct SD Card folder and flash your TFT35

makerbase-mks, 2020-01-13:
https://github.com/makerbase-mks/Datasheet/blob/master/English%20datasheet/MKS%20TFT%203.5%20V1.0DataSheet.pdf
This manual can help you.

Dikto, 2020-06-10:
What about the "Laser" interface ?
the "MKS_TFT35_laser_V1.0.2 Release file.zip" file seems to contain the interface files for 3D printing and the manual does not mention the laser part.