makerbase-mks/MKS-TFT #359: "cfg_state_state_bkcolor" element does not exist

MarshMalloW0007, 2019-12-18:
Hello, everyone.
I'm busy creating my skinpack and I found an error in the configuration file "mks_config.txt".
The "cfg_state_state_bkcolor" element does not exist in the FW while it is in the configuration file.
You must replace "cfg_state_bkcolor" by "cfg_state_colo"r so that the background color of the temperature, fan, (etc) elements can be changed.

If Makerbase needs help writing its documentation, I can help.

Sincerely
Marsh

makerbase-mks, 2020-01-13:
We will test this feature and fix it in version 1.0.3. Currently version 1.0.3 is being tested ...

MarshMalloW0007, 2020-01-13:
I can help in testing the firmware.

makerbase-mks, 2020-01-15:
Welcome to test the new firmware.
https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS-TFT3.5

closed 2021-02-24 by mks-viva