makerbase-mks/MKS-TFT #360: laser sempre ligado

evangelinosantos, 2019-12-20:
comprei a dias uma gravadora a laser equipada com a placa MKS DLC V2.0,e quando ligo os 12V na placa o laser fica sempre aceso.gostaria que me ajudassem.
desde já obrigado e desculpem mas só sei português.

marcohaddad, 2019-12-20:
Evangelino,
Talvez eu possa te ajudar apesar de nunca ter usado esse modelo de placa da
Makerbase especificamente.
Primeiramente me diga se você comprou a máquina montada ou em kit. Se for o
caso, qual kit você comprou?
Qual o laser que você está usando? Os lasers geralmente tem dois fios de
alimentação e um ou dois fios de contrôle.
Esses fios podem estar em um único conector ou em dois conectores, ou até
mesmo sem conector nenhum. Qual tipo é o seu?
Se o laser acende, muito provavelmente os fios de alimentação estão
corretos e só resta conferir os fios de contrôle. As sugestões do que fazer
depende das respostas para as perguntas acima.
Saudações,
Marco Haddad

evangelinosantos, 2019-12-21:
Boa tarde.
A máquina foi comprada na banggood,em kit, e é o kit C30,com laser de  
2.5W,tem 2 fios entre a board e a placa de contole do laser e estão  
numa única ficha.(já troquei os fios que vão para a placa de controle  
e continua na mesma)
Desde já muito obrigado e um abraço.

 

Ligações:
---------
[1]  
https://github.com/makerbase-mks/MKS-TFT/issues/360?email_source=notifications&email_token=AOC4EQDEPAUPGPWSPATHYWTQZUMEVA5CNFSM4J6BMVVKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHN6CBI#issuecomment-568058117
[2]  
https://github.com/

marcohaddad, 2019-12-22:
Tente desconectar da placa os dois fios de contrôle do laser, se isso não
apagar o laser curto-circuite os dois fios.
Se em nenhuma dessas situações o laser for apagado, então o problema está
na placa do próprio laser e não na MKS DLC.

closed 2020-04-01 by makerbase-mks