makerbase-mks/MKS-TFT #361: TFT35 with Fysetc S6

dbemowsk, 2019-12-26:
So I have an MKS TFT35 that I would like to try to use with a Fysetc S6 board. The S6 board does not have an aux1 port, but it does have a UART1 port that provides serial communication. My question is, can I just connect the +5, GND, RX and TX to the UART port to make it work or is it crucial to have the other 2 data pins from the TFT35 display connected for it to function correctly? My understanding of the pinout on the 8 pin connector of the TFT35 is this:

+5 | GND | RXD | TXD
+5 | GND | Dxx | Dyy

I am not sure of the function of the Dxx and Dyy pins on the display.

dbemowsk, 2020-01-11:
I opened this issue 17 days ago and still have had no response.  I am simply trying to find out the function of the 2 extra data pins on the main 8 pin IDC connector (not the wifi connector).  I would like to get full functionality from this display using it with my Fysetc S6 controller, but can't without knowing the operation of those two pins.

makerbase-mks, 2020-01-13:
The TX and RX pins on the MKS TFT35 are the transmit and receive pins of the serial port. You can connect to the serial RX and TX pins of your Fysetc S6 motherboard. If necessary, we can provide you with the schematic diagram of the MKS TFT35.

dbemowsk, 2020-01-13:
My question wasn't about the TX and RX pins.  There are 8 pins, 2 of them are GND and 2 are +5V.  I want to know what the two pins opposite the TX and RX pins are used for.

dbemowsk, 2020-01-25:
Can SOMEONE respond to this.  My initial post on this was on Christmas day.  This is frustrating and should be a simple response from someone that knows the function of the pins on the MKS TFT35.

makerbase-mks, 2020-06-06:
+5V  GND  RX   TX
+5V  GND  NC  NC

closed 2020-06-06 by makerbase-mks

NoPinky, 2021-04-12:
@dbemowsk 
so, have you ever made the S6 work with the TFT35? If yes, how?
Thank you

dbemowsk, 2021-04-12:
Yes. I had to make an adapter to connect the 8 pin IDC connector on the S6 to the soak port on the TFT.  You need yup connect TX on one too the RX on the other, and vice versa. Plus the 5V and GND of course.

C3P0-SF, 2021-08-14:
Can you show what this looks like? I'm assuming you used AUX 1 on TFT to the UART 4-pin on the S6 board.

dbemowsk, 2021-08-15:
C3P0-SF, here is the basic wiring needed.
[image]
 
At the TFT35 end I have an 8 pin IDC connector like this.
[image]

Then to plug into the Fysetc S6 side I just put a dupont connector for mine, but you can use one of the polarized dupont style connectors that comes with the S6 board.
[image]

Then in the firmware you need to define a second serial port in your config.h.  I don't have the firmware handy at the moment to paste in what I changed, but I'm sure you can figue it out if you read on configuring a second serial port.

C3P0-SF, 2021-08-15:
Thank you. That's what I thought. Turns out my motherboard was a dud and the serial ports didn't work

dbemowsk, 2021-08-15:
Good luck.