makerbase-mks/MKS-TFT #362: TFT35 1.0.2 crash on Pause during printing

svenddk, 2019-12-27:
TFT35 firmware 1.0.2 crashes after pressing Pause during printing, no response on touch screen buttons. After restart, screen reports "No file found". Can only work again when firmware 1.0.2 has been reinstalled from SD card.

VegetableKingdom, 2020-01-09:
Really, I have installed and used version 1.0.2 for two months and it runs smoothly.
I have also stopped printing. It was finally found that the SD card read problem

svenddk, 2020-01-10:
Thanks for your comment. Yes, really. I use a Kingston SDHC 16GB that works perfectly in all other occassions, it has been rewritten less than 100 times so it can't be flash cells worn out. The fact that your 1.0.2 pauses without problems might suggest that there is a bug in the 1.0.2 or in the wifi firmware. It pauses and resumes by Cura 4.4.1 through wifi, just not resumes bu touch screen

closed 2020-03-16 by svenddk