makerbase-mks/MKS-TFT #364: The MKS TFT35 is unable to control the heating of the camera.

Error26, 2020-01-09:
The MKS TFT35 is unable to control the heating of the camera. Is it possible to do this in some way?

VegetableKingdom, 2020-01-09:
camera？

Is it because the communication between the motherboard and the screen is abnormal?
Baud rates are inconsistent?

Error26, 2020-01-09:
I don't know how to translate correctly. In Marlin's code is this  function.
#define TEMP_SENSOR_CHAMBER 1
#define TEMP_CHAMBER_PIN   P0_25_A2   // Analog Input A2 (TH2)

VegetableKingdom, 2020-01-13:
First you need to confirm whether the thermal model you use is the same as the model you set.
You can connect the PC with a USB cable without using the touch screen first to see if the temperature can be displayed normally. If the host computer software can detect the temperature, but tft35 cannot, it means that the baud rate of the motherboard and tft35 is different. You need to reset the baud rate to update the firmware (both the motherboard and the screen are OK), keeping the motherboard and the screen consistent
If the PC does not read the temperature, it means that the thermal type set by the firmware is inconsistent with the model used, or the temperature measurement port of the motherboard is broken, causing the motherboard to fail to recognize it.
The tft screen only obtains temperature through commands. The real temperature measurement function is performed on the motherboard.

Error26, 2020-01-13:
> Спочатку потрібно підтвердити, чи використовується тепловимісна модель, яку ви встановили.
> Ви можете підключити ПК за допомогою кабелю USB, не використовуючи спочатку сенсорний екран, щоб побачити, чи може нормально відображатися температура. Якщо програмне забезпечення хост-комп'ютера може визначати температуру, але tft35 не може, це означає, що швидкість передачі даних материнської плати та tft35 відрізняється. Вам потрібно скинути швидкість передачі даних для оновлення мікропрограмного забезпечення (і материнської плати, і екрана в порядку), підтримуючи сумісність материнської плати та екрана.
> Якщо ПК не зчитує температуру, це означає, що термічний тип, встановлений прошивкою, не відповідає із використовуваною моделлю, або порт порту для вимірювання температури материнської плати зламаний, через що материнська плата не розпізнає її.
> Екран tft отримує температуру лише за допомогою команд. Функція вимірювання реальної температури виконується на материнській платі.

MKS TFT35 and others send command "M141 S" and "M191 S" with G-coge to MKS board. But on display MKS TFT 35 i dont see and dont can control the temperature of hotting chamber. On PS  and the other display "LCD 12860" i see the temperature and can control the temperature.

closed 2020-01-13 by Error26