makerbase-mks/MKS-TFT #365: Mks 32 doesn't communicate with ramps 1.4

FearCapone, 2020-01-10:
My mks 3.2 v4 doesn't work with my ramps 1.4 connected on aux 1.
Booth the same baudrate, latest firmware. 

If I put the mks tft on a ftdi adapter I can see the commands like G28 if I touch there but it takes many time.
The mks send's M105 every few seconds.

No temperature on display, no motor movement possible

makerbase-mks, 2020-01-13:
The screen can send M105, indicating that the screen and the motherboard can communicate. If the PC is connected to the motherboard, can the host computer work normally? Which version of Marlin does your motherboard firmware use?

FearCapone, 2020-01-14:
Yes the host can work normally.
I found the problem, it's the Ch340 USB to serial ic. 
Removed it and replaced it with a ftdi (rx, TX, dtr, gnd & 5v)
Mega with 16u2 or similar USB to serial ic works out of the box.

FearCapone, 2020-01-14:
Solved

closed 2020-01-14 by FearCapone