makerbase-mks/MKS-TFT #367: MKS TFT35 bugs

Error26, 2020-01-11:
Found some bugs in the firmware.
1) Incorrect display of auto leveling icon and lack of signature.
```
#leveling mode(manual:0; auto:1; conceal leveling button:2)
>cfg_leveling_mode:1
```
[image]

2) After the error is displayed a picture with the letter G.
[image]
[image]

makerbase-mks, 2020-01-13:
We will fix this problem in version 1.0.3, and the firmware of version 1.0.3 is currently being tested.

Error26, 2020-01-13:
> Ми вирішимо цю проблему у версії 1.0.3, і в даний час тестується прошивка версії 1.0.3.

I can help in testing the firmware.

makerbase-mks, 2020-01-13:
Welcome, we have several issues to be fixed, which will be released soon

MarshMalloW0007, 2020-01-13:
Hello,

When cfg_leveling_mode is set to 1, the bmp_leveling.bin icon is automatically replaced by the bmp_custom1.bin icon.
In the firmware icon pack, bmp_custom1.bin does not comply with the graphic charter.

Sincerely
Marsh

MarshMalloW0007, 2020-01-13:
Hello,

When cfg_leveling_mode is set to 1, the bmp_leveling.bin icon is automatically replaced by the bmp_autoleveling.bin icon.
In the firmware icon pack, bmp_autoleveling.bin does not comply with the graphic charter.

Sincerely
Marsh

Error26, 2020-01-13:
Can you provide your configuration file? I'm try to help you.

fnhhnxfn, 2020-02-02:
hi from where can i get this skin pack