makerbase-mks/MKS-TFT #368: Customization of screens and background

MarshMalloW0007, 2020-01-13:
Good morning all.
I am busy creating my skinpack for a TFT 35 and I have a problem with a background.
All the parameters in the mks_config.txt for the backgrounds are set to white.
Also, the text for the Z remains white.
[image]

Sincerely
Marsh

Error26, 2020-01-13:
Can you provide your configuration file? I'm try to help you.

MarshMalloW0007, 2020-01-13:
#---------------------------------------------------------------------------------
##### Display Customization(button and logo customized see User's Manual) #######

#the color of
# screen background
>cfg_background_color:0xFFFFFF
# title text
>cfg_title_color:0x000000

#the color of temperature, fan,(etc) 
# background
>cfg_state_bkcolor:0xFFFFFF
>cfg_state_color:0xFFFFFF
# text
>cfg_state_text_color:0x000000

#the color of file
# button
>cfg_filename_bkcolor:0xFFFFFF
# text
>cfg_filename_textcolor:0x000000

#the color of general button
# background
>cfg_btn_bkcolor:0xFFFFFF
# text
>cfg_btn_textcolor:0x000000

#the color of status button
# background
>cfg_state_btn_bkcolor:0xFFFFFF
# text
>cfg_state_btn_textcolor:0x000000

#the color of "back" button
# background
>cfg_back_btn_bkcolor:0xFFFFFF
# text
>cfg_back_btn_textcolor:0x000000

#the color of selected button
# background
>cfg_sel_btn_bkcolor:0xFFFFFF
#the color of selected button
# text
>cfg_sel_btn_textcolor:0x000000

#the color of dialog button
# background
>cfg_dialog_btn_bkcolor:0xFFFFFF
# text
>cfg_dialog_btn_textcolor:0x000000

makerbase-mks, 2020-01-14:
This is a firmware issue, I can fix it today and upload new firmware.

makerbase-mks, 2020-01-15:
Welcome to test the new firmware.
https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS-TFT3.5

closed 2020-05-27 by makerbase-mks