makerbase-mks/MKS-TFT #371: MKS TFT70 not connect Wifi and don't support 3 extruders

biroli3d, 2020-01-15:
Greetings, I'm using a TFT70 and I tried many times setup "mks_config.txt" and I couldn't connect. I try with MKS WIFI TFT and MKS HLK-WIFI (I have both hardwares) and nothing. How I solve it?
I have a THC-01 (3 in 1) and I'd like to setup at menu all extruders, now I'm using the command "more buttons" and there I cannot setup a lot.
Last week I updated "cura" and there have some problems to create the preview, and I need to use the older version.
Any help would be appreciated. Thanks.

makerbase-mks, 2020-01-16:
Hi!
1. For the problem that MKS TFT cannot connect to wifi, can you send the configuration file here?
2. For 3 extruder, we are evaluating development…

biroli3d, 2020-01-18:
1. Thank you very much, I changed the configuration and hardware and connected very well.
2. I'm waiting the next update about this.
In next update it's possible create a setup for you write text bellow the icons? Like as "custom" language?
This would be possible you have all languages in your screen, without worrying about developing so many languages.