makerbase-mks/MKS-TFT #372: MKS TFT 28 Marlin 2.0 Ramps 1.6 not working

Rememberthecreed, 2020-01-16:
so a while ago, I purchased a Mks Tft 28 screen to use on my 3d printer. I replaced my Melzi board with a Ramps 1.6 + with and Arduino Mega combo (I plan on replacing it with a re-arm board at a later date, but it's what I had on hand.) After flashing on the latest update of Marlin 2.0, and using the 3.0.2 software update from the MKS TFT file, I plugged the tft into the aux 1 on the ramps board. It powered up, and the screen worked fine, but non of the commands moved my printer. None of the thermistors register either. The printer moves just fine when connected to Repetier. But the screen won't control the printer (I have tried it both plugged into my laptop and unplugged, but it doesn't make any difference.) 

both baudrates in Marlin and on the TFT are 115000 
Serial port is 0 (I don't see where to change the serial port in the tft firmware, so it's 0 in Marlin)
serial port 2 is uncommitted (Marlin won't compile with the port commented)    

I've seen many video's of the screen working just fine on ramps 1.4 boards, but not many on the 1.6+, but I don't think there is much difference, as the pinnouts are the same for aux 1.

makerbase-mks, 2020-01-16:
Hi! The problem you feedback should be caused by incompatible interfaces.
1. AUX-1 interface definition, MKS Gen_L V2.0
[image]
2. RAMPS1.6 interface is defined as follows, I think this is the reason
[image]
Therefore, the serial cable of TFT28 is connected to A3 and A4.the serial cable of TFT28 is connected to A3 and A4. Therefore, you need to skip the A3, A4 side and directly connect the pins on the D1, D2 side.

Rememberthecreed, 2020-01-17:
I did as you suggested and still no dice. I made two 4 wire DuPont connectors, and wired it as suggested, but the mks still won’t connect to the board.

Rememberthecreed, 2020-01-25:
After quite a bit of monkeying around with code and wires, I've given up and just ordered an MKS Gen L. Works just fine now. I'm under the impression that there's some kind of pin assignment I missed from the 1.4 to the 1.6. Either way, thank you for your time.

closed 2020-01-25 by Rememberthecreed