makerbase-mks/MKS-TFT #373: Manual mesh bed leveling process?

dexx086, 2020-01-17:
I've a TFT28 and an MKS Gen L v1.0 motherboard. I tried, but cannot make manual mesh bed leveling working on the TFT.

The motherboard (latest Marlin 2.x) is configured properly, with the stock LCD it's working.

On the TFT the leveling menu has only Point1, Point2 (or First, Second) options. If I press the first point, it's OK, it moves to the first bed leveling point. But now what should I do? How can I modify the Z height to set it up, how to use babystepping, and how to save the current Z height and move to next point???

On stock LCD, manual leveling homes XYZ, automatically moves to 1st point, shows a babystepping 'Move Z' menu for each point, and after click, it saves, goes to next point, and guides the user through the whole process.

How is manual mesh bed leveling process intended to work with MKS TFT28?

dexx086, 2020-01-23:
No response yet on this issue, does it mean MKS is not willing to support a method as basic as manual mesh bed leveling?

fireflylr, 2020-03-06:
Tft28's leveling is different from other ordinary LCD screens. If you want to use the manual leveling function of tft28, you need to adjust the mechanical structure of the printer to complete the leveling step. If you want to use the automatic leveling function of tft28, you need to add additional sensors
----mks

dexx086, 2020-03-06:
Manual mesh bed leveling doesn't equal to manual leveling the bed with the hotbed's screws... Manual mesh bed leveling is much more than physically adjusting the hotbed itself to be parallel with your "horizontal" plane.

Mesh bed leveling takes measures (by you!) on different points of the hotbed, and Marlin can use that to **continuously correct the Z-height**, according to your manually measured "mesh". This way, even a warped bed can be corrected and used! With simple manual (screw) bed leveling, not.

And to be honest: as all mesh bed level handling functions are available through gcode commands, it wouldn't be a big task to implement it. Just should get the number of X/Y points, cycle through all of them, allowing babystepping meanwhile, and setting the values for each point. All these with simple gcode commands. Not a big deal...

But I don't think MKS would take the precious 1-2 day of development into this... I saw much more problems regarding their firmware. Especially the attitude how they care about it, and secondly for not having it open source...
These are the reasons I don't even use this TFT... I bought it for nothing, went back to simple character based stock LCD...

I think the hardware itself is OK, the only case I will be able to use it if Marlin would handle graphical LCD's by its own, only for **displaying**, but the LCD should not have it's own firmware... The LCD should only be a 'monitor', not a full-blown controller.

All such custom LCD firmwares will lack many-many features that Marlin offers, so I don't see any reason to have a +1 firmware running on an LCD, being behind Marlin's functionality, and adding a +1 fail factor to your system. I'm a disappointed MKS buyer.

KixMan28, 2020-04-18:
@dexx086 I agree with you 100%. I was looking for this functionality on my TFT28, but as I couldn't find it I looked around here, maybe someone else found a solution for it, but I am sad to see that there is no solution for it.
Actually I can't make it work together with a BigTreeTech SKR1.3 mainboard. I If I connect the display to the mainboard on the AUX1 port I can have two way communication, but it won't provide power to it (won't lit up). In order to have it showing something I need to power the TFT28 externally with 12V.

KixMan28, 2020-04-30:
@fireflylr Now that the babystepping is enabled in the TFT firmware, it would be nice to add the manual mesh bed leveling to the firmware. The Gcode for the manual mesh bed leveling is G29 and you can read about it here: [https://marlinfw.org/docs/gcode/G029-mbl.html](url). 
Let us know if you plan to implement this?

KixMan28, 2020-05-03:
@makerbase-mks Now that the babystepping is enabled in the TFT firmware, it would be nice to add the manual mesh bed leveling to the firmware. The Gcode for the manual mesh bed leveling is G29 and you can read about it here: https://marlinfw.org/docs/gcode/G029-mbl.html.
Let us know if you plan to implement this?

KixMan28, 2020-05-13:
@makerbase-mks @fireflylr Any reply here?

dexx086, 2020-05-13:
I don't think they'll care about it... Almost 5 months elapsed since my initial question, and didn't get any meaningful or at least promising reply...

Btw: I have an unusued TFT28 to be sold, anyone interested? :D

karabas2011, 2020-06-03:
You can try create two custom button G29 S1 and G29 S2, but you need to jump to move menu  after every press.

karabas2011, 2020-07-07:
Ok, I have implemented manual MBL using example code in 1.0.7.  Works but only english now.
Still struggle with  Zoffset ( M420 V or M290  insteed of M851) when leveling on. I want to see current value on dsplay.

simpat1zq, 2020-10-17:
> Ok, I have implemented manual MBL using example code in 1.0.7 on TFT35. Works but only english now.
> Still struggle with Zoffset ( M420 V or M290 insteed of M851) when leveling on. I want to see current value on dsplay.

@karabas2011 Could you please share what you did to enable manual mesh leveling?

edoluck90, 2020-12-23:
> 
> 
> Ok, I have implemented manual MBL using example code in 1.0.7 on TFT35. Works but only english now.
> Still struggle with Zoffset ( M420 V or M290 insteed of M851) when leveling on. I want to see current value on dsplay.

Mate can you share your solution ?

Patrick-81, 2021-02-10:
@karabas2011 It would be nice to share the way to enable the MBL. Thank you.

bart-tech3d, 2021-03-16:
Hi, I have added a comfortable control for MBL in the form of new icons. see the download link https://github.com/bart-tech3d/MKS-TFT28-32 where the modified fw is. Alternatively, just change the mks_config.txt configuration file according to your requirements.
[image]

karabas2011, 2021-03-17:
> @karabas2011 It would be nice to share the way to enable the MBL. Thank you.

Unfortunatly I damaged code for Print and jumped to BTT testing. 
I tried BTT version for MKS_TFT35 made by Darksprite 
It works but touchscreen  has dead zones. And i2c eeprom does not save

karabas2011, 2021-03-17:
> Hi, I have added a comfortable control for MBL in the form of new icons. see the download link https://github.com/bart-tech3d/MKS-TFT28-32 where the modified fw is. Alternatively, just change the mks_config.txt configuration file according to your requirements
.
Did you mod the PlatformIO version of firmware?

bart-tech3d, 2021-03-17:
Hi, I don't have the source code, he's probably nowhere. I did it's simply.
I did this by uploading a .bin file to https://hexed.it/ for view hexadecimal and changing the name for the MBL icon there, otherwise it could not be changed. In retro-style mkstft28.bin file changed more to mbll - number of characters preserved. Then save. When changed in the text editor, the code is corrupted and cannot be uploaded to the display.
Other commands are already simple in the form of their own buttons and commands, see. excerpt from mks_config.txt below:
#set number of  "More" button 
>moreitem_pic_cnt:6

#edit command for 1~7 "More" button,each command must be separated by semicolon ";"
>moreitem_button1_cmd:G91; prepeare for MBL (incremental)

>moreitem_button2_cmd:G29 S1; start to first point

>moreitem_button3_cmd:G29 S2; move to next point

>moreitem_button4_cmd:M500; save to EEPROM

>moreitem_button5_cmd:G1 Z0.01 F500; move Z +0,01

>moreitem_button6_cmd:G1 Z-0.01 F500; move Z -0,01

>moreitem_button7_cmd:
[image]

karabas2011, 2021-03-17:
> I don't have the source code, he's probably nowhere
it's open source. You can download it. Last version even migrated to Platformio

bart-tech3d, 2021-03-17:
karabas2011 could you write a link ? Thanks

bart-tech3d, 2021-03-17:
karabas2011 probably I find it. its OK.

bart-tech3d, 2021-03-17:
karabas2011 no I canot it find . It was some other.

bart-tech3d, 2021-03-17:
matmcfly  I haven't drilled in TFT35 yet and I don't know what the menu structure is. But it should go the same way. Or just edit it in mks_config.txt and use the "more" icon for MBL.
I'm sorry, but I haven't had much time for that lately.

karabas2011, 2021-03-18:
> karabas2011 could you write a link ? Thanks

https://github.com/makerbase-mks/MKS-TFT35-Firmware

matmcfly, 2021-03-20:
@bart-tech3d please send me mesh bed leveling icon thanks

bart-tech3d, 2021-03-20:
@matmcfly  you can download it here https://github.com/bart-tech3d/MKS-TFT28-32/raw/main/MBL_ICON.zip

closed 2022-07-07 by dexx086