makerbase-mks/MKS-TFT #374: Long filenames issue

dexx086, 2020-01-17:
I have TFT28, and I find very hard to find the files I want to print. Currently the firmware shows the filename in 2 rows, 8 characters in each row. I have longer filenames... And usually I have the "version" information at the end of the files, which is not visible at all on the TFT.

And the most annoying thing: if I (kind of blindly) click on a file and I see it's not the one I want to print (at least before print, the full filename is visible on the confirmation page, that's at least a checkpoint), and I click Back button, it goes back to the root folder!!! At least, should stay at current folder, so I could click on the next file, maybe that's the one I want to print... It's very annoying.

Couldn't the file browser at least scroll long filenames?
Or, would be more convenient, show a simple full-width list of filenames? Possibly all filename would fit in the display at full-width. It could be an option to show file browser in 'icon' mode, or in 'simple list' mode.

Any chance to make the file browser more usable in any of the proposed ways?

makerbase-mks, 2020-01-17:
We will try this feature in new software version.

dexx086, 2020-01-17:
Thanks for your quick and promising response.

closed 2020-06-06 by makerbase-mks