makerbase-mks/MKS-TFT #376: MKS TFT35 M0/M01 commands not working.

visibl3ghost, 2020-01-18:
I have prints that require multiple filament changes, and I want to use Cura's 'Pause at height' gcode modifier. This inserts a M0/M01 command info the gcode itself. 

Currently, when the TFT35 encounters this command it pauses the printer for a short period of time (<1second) and then continues on printing.

The only way for me to time my filament changes currently is to wait and manually pause the print and the correct time.

Can M0/M01 please be fixed so the TFT35 stays in a paused state?

Thanks!

rndsum, 2020-01-19:
You using the TFT in SPI mode or actual LCD cables in emulator mode? 

http://marlinfw.org/docs/gcode/M000-M001.html

> Without an LCD controller or EMERGENCY_PARSER this command is ignored.

visibl3ghost, 2020-01-20:
You were right, I enabled EMERGENCY_PARSER and now the printer will pause, however there is no way to resume print.

What I mean is, when the printer encounter the M00 command it pauses, but the TFT35 does not indicate that it has paused, and there is no way to resume printing.

nehu92, 2020-01-22:
same problem it is really anoying not to have that command implemented in the tft. The best option for now is to use the pc :,(