makerbase-mks/MKS-TFT #379: MKS-TFT35 status screen

dereik, 2020-01-28:
is there any way to get an info / status display fpr the TFT35.
i would like to see the bed an hotend temp at the same time.
for the TFT32 there is an retro firmware but not for the TFT35

makerbase-mks, 2020-03-12:
Hi! Currently, there is only one style of MKS TFT35, and we will collect more needs and evaluations.

gizmocuz, 2020-04-10:
I would love to see this option too ! This should be the default home screen after bootup
With a Abort/Stop button would be nice

I think this is the same as this issue

https://github.com/makerbase-mks/MKS-TFT/issues/292

If possible please reopen that one

tschork, 2021-02-03:
Yeah, that would be nice to have.
I just installed my tft35 today and was surprised not to find a screen like this.

closed 2021-02-24 by mks-viva