makerbase-mks/MKS-TFT #381: Boot loop after second flash

rlayne, 2020-02-02:
First flash went ok on the MKS3.2 v4.0 via sd card, retro, win8
The wifi chip wasn't installed at the time though, so I installed it, and on next power on the wifi firmware updated.  Worked great, the only problem at this point was the wifi didn't take the settings from the config (i.e. I wanted station mode not AP mode) so I decided to reflash thinking that this would fix the wifi settings.

I renamed the original files on the SD card back, and recycled the power.  Boot loop.
Happens every time I try to flash anything now despite taking the original steps.

Does anyone have any thoughts?

fnhhnxfn, 2020-02-02:
hi its pretty simple to set wifi with tft 
first install mks wifi module
then in txt file enable TFT WIFI and select ap or sta it depends on your choice weather you want to connect via router or with hotspot 
if selected ap then put your wifi name and password  
then copy mks32.txt and mks28.bin to your sd  card and power on the printer

rlayne, 2020-02-02:
> hi its pretty simple to set wifi with tft
> first install mks wifi module
> then in txt file enable TFT WIFI and select ap or sta it depends on your choice weather you want to connect via router or with [...]

Sorry if it wasn't clear enough, but that is precisely what I was doing.

rlayne, 2020-02-02:
I've found a workaround.  As mentioned regardless of the settings specified for station mode in the config file, the board wifi reverts to default AP mode no matter what.

However, connecting to the TFT in AP mode via a browser reveals the usual config screen, and setting the WIFI values here _does_ stick after a reboot.  

Closing - hope this is useful to somebody.

closed 2020-02-02 by rlayne