makerbase-mks/MKS-TFT #382: TFT 35 issue

fnhhnxfn, 2020-02-02:
Hi Guys ,
I know its a bit strange but i tried to flash Btt tft 35 firmware to mks tft 35 bcz its open source but my screen has gone black and now i am not even able to flash mks firmware to it i checked the hardware ,hardware is working fine 
do anyone know that how can i erase the old firmware and flash the new one?
MKS plzzzz help me

makerbase-mks, 2020-02-11:
The BTT firmware and MKS TFT are incompatible and temporarily unavailable.
More information about MKS TFT, we will further open up

darkspr1te, 2020-02-14:
see https://github.com/darkspr1te/BIGTREETECH-TouchScreenFirmware for working firmware for tft35v4 , easy to port to others, also opensource bootloader in case your's is damaged. Easy to flash if you have a usb - serial or arduino laying around.

UnderXirox, 2020-02-18:
Hello, here are several months
Black screen also on your Mks TFT3.5 screen
a quick solution to solve the problem?

release source code

makerbase-mks, 2020-02-20:
Hi，user-defined button display requires the following configuration
#set number of "More" button
moreitem_pic_cnt:0

bthome, 2020-04-01:
@fnhhnxfn Did you ever get your issue resolved?  I'm in the same boat.  I would like to get back to stock bootloader and firmware, can't find any good instructions or file locations anywhere.

karabas2011, 2020-04-24:
Get Bootloader.hex for you mks_tft version ( my is v1.0 )
flash it with ST-link ( or maybe other ways exist) 
Now your display starts to booting.

bthome, 2020-04-24:
> Get Bootloader.hex for you mks_tft version ( my is v1.0 )
> flash it with ST-link ( or maybe other ways exist)
> Now your display starts to booting.

Can you point me to where a download would be?  Or extract yours and post it?

darkspr1te, 2020-04-29:
Good News Everyone, We have a almost working bootloader for MKSTFT35 V1 boards, the same code used to "enable the parts of the cpu" is the same code i need to add to BTT builds and Marline touch to enable those users to have a choice. I am tho asking if we could crowd source a real unit for me to keep up progress, once am done doing the bootloader and main port of Marlin touch my access to the device will be limited as my test user will be using the device, having one myself will ease that issue. Send me direct message if you wish to assist. 
Well, onto unknown code and Beyond !!

karabas2011, 2020-04-30:
> > Get Bootloader.hex for you mks_tft version ( my is v1.0 )
> > flash it with ST-link ( or maybe other ways exist)
> > Now your display starts to booting.
> 
> Can you point me to where a download would be? Or extract yours and post it?

[mkstft35_v1.0.3_bootloader.zip](https://github.com/makerbase-mks/MKS-TFT/files/4556765/mkstft35_v1.0.3_bootloader.zip)

karabas2011, 2020-04-30:
> Good News Everyone, We have a almost working bootloader for MKSTFT35 V1 boards, the same code used to "enable the parts of the cpu" is the same code i need to add to BTT builds and Marline touch to [...]

How do you support MKS TFT35 ?? It is  stm32f407, not stm32f107 as bigtree one

darkspr1te, 2020-04-30:
another user has connected his st32f407ve display to a pc and has let me develop the code for it via discord/remote access. I have written a new bootloader for it without actually ever having had one, not easy as doing test code for a device you cannot see/touch is not easy. Thankyou for posting the bootloader, this will help us finish up the open source bootloader we have been working on.

trucutrubl, 2020-09-01:
> link

Hi,
How should I connect a st link to a mkstft35v1 tft screen? Is there any diagram?

Thanks and regards.

darkspr1te, 2020-09-01:
on the back of the screen is a connector labled JTAG, its pins are 
1 SWDO
3 SWCLK
5 RESET
2 3.3v
4 gnd
6 gnd
you only need to connect swdio/swclk to the stlink adapter, the power comes from the printer board as normal.  diagrams can be found at [https://github.com/makerbase-mks/MKS-TFT-Hardware/blob/master/MKS%20TFT35%20V1.0_003/MKS%20TFT35%20V1.0_003%20SCH.pdf](url)

karabas2011, 2020-09-01:
In my case it works only with 5v not 3.3.
After reflashing I use usb3.0 connect to check without printer. ST-link does not provide enough current for LCD to work properly

UnderXirox, 2020-09-01:
Hello thank you
for your work 👍 i finally got 1 beep from the lcd tft.35
MKS but still oas of lcd on black screen
pardon my English

available to help I wing mks tft3.5v1.0 a digital osilloscope, stlink V2 for debug flashing remains to have a small analog analyzer

once again sorry for my english

UnderXirox, 2020-09-01:
Yes Thank you for your bootloader it works for me on my MKS TFT3.5V1.0 flash or Save Via STlinkV2

Thank you

UnderXirox, 2020-09-01:
pin type MKS TFT3.5V1.0  Jtag

PIN Jtag MKS TFT3.5V1.0
_______________________
| 3.3      GND    GND  |
|   °          °          °      |
|   °          °          °      |
|SWIO   SWLK   RST   |
|______________________|

Fasher with STlinkV2:
[mkstft35_v1.0.3_bootloader.zip](https://github.com/makerbase-mks/MKS-TFT/files/5156616/mkstft35_v1.0.3_bootloader.zip)

here is my Dump once the mks3.5v1.0  unbrick

[Dump Mks_tft35_v1.0.3_bootloader.hex.zip](https://github.com/makerbase-mks/MKS-TFT/files/5156601/Dump.Mks_tft35_v1.0.3_bootloader.hex.zip)

thank you for your work

darkspr1te, 2020-09-08:
just a update on the BTT port to the MKSTFT35,
[image]
many fixes to get this far. Working so far is LCD,SDCard, USB, SPI (for images/icons) , i2c (for parameter storage), USART1/3 (for wifi/marlin) , touch is very buggy atm due to use of a different chip, without refactoring BTT's code a great deal I have yet to fix this bug. but progress is being made

darkspr1te, 2020-09-10:
it's now live and usable - still some minor bugs but core support is ready for MKSTFT35V1_0 (or any stm32f4xx board) 
https://github.com/darkspr1te/BIGTREETECH-TouchScreenFirmware

UnderXirox, 2020-09-10:
Thank you I test this tomorrow

do you want me to check something in particular?

darkspr1te, 2020-09-10:
Report any bugs you find (via darkspr1te/bigtreetech repo) , new bootloader coming soon to be more compatible with both mks/btt firmware in one package.

darkspr1te, 2020-09-10:
open source bootloader supporting original firmware and btt firmware found at 
[https://github.com/darkspr1te/mkstft35_V1_open_bootloader](url)

jcbr92, 2022-11-05:
> Hello I repeat the symptom of my concern resolved thanks to you
> 
> So one day by plugging in my stlink_v2 (Original not the clone) A TFT mks3.5v1.0 A dirty manipulation on my part and op brick of the lcd Black screen without beep at boot or reading the SD card Control with Ocsilloscope nothing at all flat trace And there is a possibility to take my hand on it with a Serial usb convert in short circuit the resistance R10 and GND Nothing nada My concern was the bootloader which was solved with
> 
> https://github.com/makerbase-mks/MKS-TFT/files/5156616/mkstft35_v1.0.3_bootloader.zip
> 
> Thank you I test this tomorrow
> 
> do you want me to check something in particular?
> 
> sorry for my english.

can you help me please?
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/discussions/2627