makerbase-mks/MKS-TFT #384: MKS-TFT70

ant0nyk1ng, 2020-02-05:
Firmware version 103

When the filament sensor is used the screen freezes. To start printing again I have to turn the printer off then back on and the resume works and it continues.

[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/4157955/mks_config.txt)

VegetableKingdom, 2020-02-15:
Is it possible that the set trigger_Level is the opposite? That ’s why printing is paused at the beginning.

ant0nyk1ng, 2020-02-16:
The printer worked fine with version 101. 

The Print starts and prints perfectly till the filament runs out. The print pauses correctly but when I go to change the filament the screen is frozen.

I turn the power off then on. It starts and askes to resume print. I do, then I can change the filament and resume printing.

makerbase-mks, 2020-02-25:
We will test this issue and update a firmware version as soon as possible.