makerbase-mks/MKS-TFT #385: MKS TFT70 touch screen cannot use to control

neonaing, 2020-02-23:
The touch screen has an issue about activate, touch one button it activate others too. I reported to the seller and he replied as follow,
" Sir, because the screen used by tft70 is a resistive screen.Therefore, it may cause a false touch during the touch process.
There may be reasons for this.
The area of the touch screen is too large. You can try to touch it with your fingertips or something with a small contact area, which may reduce the chance of accidental touch.
In another case, is it possible that the metal frame on the screen is pressed tightly and caused a wrong touch.
See if this is the cause. "

I took his advise with loose the frame and use my old PDA's stylus, still the same problem. I think it is a calibration problem, and ask him how to re-calibrate. He reply as follow

" Hello sir, we will release the new tft70 firmware in March. Our engineering team will open the new firmware and screen calibration function. At that time you will use the new firmware to update and recalibrate the screen.
Should reduce the chance of your screen being touched by mistake '

Hopefully that new firmware will resolve the issue otherwise the screen is useless.

I made a video on youtube and youku.

[Youtube](https://www.youtube.com/watch?v=HL-OISJkcr8&feature=youtu.be&spm=a2g0s.imconversation.0.0.5fc53e5f6RQ1UP)

[Youku](https://v.youku.com/v_show/id_XNDUzOTM0MTMzMg==.html?spm=a2hzp.8244740.0.0)

Thanks

makerbase-mks, 2020-02-25:
Hi, we will add screen calibration function in the next firmware version

neonaing, 2020-03-02:
After updating the v1.0.4 and enable calibration, the problem still there. Now it is different, before it activate randomly. Now it activate with the last one. ie 

if I click x+ first, second press x- will also activate x+ , third click y+ will also activate x-.
like this  
start with **x+**......**x-** _x+_.....**y+**_x-_......**z+**_y+_......**z-**_z+_........**y-**_z-_  and so on... 
wheres Bold ones are I click, italic one activate together with bold one
in this sequence if I double tap, it only activate the button. If I press other button the sequence start again. 

 I can't use the LCD to control the printer.