makerbase-mks/MKS-TFT #387: about baby-stepping

Bradley-Liu, 2020-03-05:
i am using a mks tft35 with the 1.0.4 firmware. i noticed that you have added a babystep interface. I want to make some recommendations. The babystep interface can't display the current z-offset value, rendering it very inconvenient to modiy the value. I don't know where i am at. Also the babystep interface don't have a 0.01mm or 0,25mm step options. The smallest is 0.1mm. I think it's too large for a babystep to work.

makerbase-mks, 2020-03-05:
Thanks for your suggestion, we will improve it

Bradley-Liu, 2020-03-10:
Thanks for you prompt action. I've updated to the 1.05 version firmware. Now babystepping is working perfectly.  However, i still got one recommendations for you. In tools->more->z-offset settings, the firmware still got z-offset displayed at 0. It is inconsistent with the babystep setting. I hope you can fix that in the following firmware. Thansk!

makerbase-mks, 2020-04-01:
You can refer to it: https://github.com/makerbase-mks/MKS-TFT/issues/389

closed 2020-04-01 by makerbase-mks