makerbase-mks/MKS-TFT #388: About the wifi print.

Bradley-Liu, 2020-03-05:
I've got wif module for tft35 and i installed a mks wife plugin for cura4.4. After i sliced the model and send it the printer, my start gcode is missing from the gcode file. The printer start printing without preheating and homing. The wif print function is of no use if i can't add start gcode set in my cura printer profile. Could you please modify the plugin settings to make start gcode work.

makerbase-mks, 2020-03-05:
Hi! The mks wifi plugin is just a plugin and cannot modify cura itself. This part can only be, when you slice, increase the corresponding starting gcode and ending gcode.

Bradley-Liu, 2020-03-05:
where can i add the start gcode ? I think the plugin can modify the gcode file before transfering it to the printer, or you can modify the plugin to give use an interface to add their customized start and end  code. i think that will  make the end user experience of the sceen and wifi module even better.

Bradley-Liu, 2020-03-05:
i did a test. i upload the gcode through wifi and open the file transfered to the sd card and noticed that the preheating and homing gcode was in the  file. But, i don't know why the printer omit that and begin to print the first layer. Will the screen modify the gcode before send it marlin? I am really confused.

makerbase-mks, 2020-03-12:
Hi! You can use cura to add the start gcode and end gcode when setting up the machine.MKS TFT just sends instructions without modifying gocde. so it's added in the wrong position?