makerbase-mks/MKS-TFT #390: TFT28 colours not correct?

flygja, 2020-03-12:
I'm using the red icon pack and retro firmware. When I set the font colour to 0xxFFFFFF, I don't see the labels for the buttons. When I set the status font colour to 0xFFFF00 for yellow, I get blue instead. I have attached screenshots and my config file. I'm using firmware 3.0.5

[image]
[image]

[mks-config.txt](https://github.com/makerbase-mks/MKS-TFT/files/4324097/mks-config.txt)

DenisShelema, 2020-03-13:
RTFM. This displays use BGR not RGB colors. 0x00FFFF for yellow.

flygja, 2020-03-16:
The manual says yellow is 0xFFFF00. Still doesn't explain why the button labels are missing. They are missing even when text colour is set to white 0xFFFFFF.

[image]

DenisShelema, 2020-03-16:
Nice joke from our friends =))). I can't see cfg_btn_text_offset parameter in MKS-TFT28 default config, like in MKS-TFT3.5 config. Is MKS-TFT28 firmware supports this parameter? If cfg_btn_text_offset value is too high or too low button text becomes invisible.
BTW, I think there is no solution for this problem, because [majurka](https://github.com/majurca/MKS-TFT28-NEW-PICTURES) theme have text embedded in pictures. Pretty sure this is for good reason.

flygja, 2020-03-16:
Nope, this is not in the config file for TFT28. I think when I change any "background color" option to anything except black, the text labels disappear. If I change it back to black, then they all appear again.