makerbase-mks/MKS-TFT #394: TFT28 touch calibration issue

flygja, 2020-04-08:
Hi, how can I re-calibrate the touchscreen. It was working for a few weeks and now the right side calibration is not accurate. I can't press the "print" or "back" buttons for example. Please see attached video.

[20200407_202749.zip](https://github.com/makerbase-mks/MKS-TFT/files/4447935/20200407_202749.zip)

makerbase-mks, 2020-04-08:
Hi! First of all, I want to confirm whether your TFT28 is a framed version, and check if there is any other touch area that presses be the screen.

flygja, 2020-04-08:
Its framed version - TFT28 v4.0. Seems ok... the frame is not touching the screen. Seems like whole screen calibration is off. I'm using win 8 theme. When I press "print", it goes into "home" menu. When I press "home", it goes into "level" menu. When I press "set", it goes into "level" menu. 

[image]

flygja, 2020-04-11:
Any help?