makerbase-mks/MKS-TFT #395: Building the project

cosminroger, 2020-04-14:
Hi.

I would like to add more functionality to the TFT28/32 model but I don't know how to build it to be suitable for the screen. Do you have a tutorial or step by step guide somewhere?

I hope I will hear from you soon,
Roger

el-quique, 2020-04-15:
equally. i need tutorial

el-quique, 2020-04-15:
Maybe this work:
https://github.com/robotsrulz/MKS-TFT/issues/4#issuecomment-311905079

cosminroger, 2020-04-16:
I have looked there but unfortunately is not what I need. I want to build it a .bin file and upload the new firmware through the SD Card.

el-quique, 2020-04-17:
> I have looked there but unfortunately is not what I need. I want to build it a .bin file and upload the new firmware through the SD Card.

Hi view this link, working for mi TFT clon MKS in Artillery X1:
https://github.com/Blueforcer/MKSTFT_Marlin_Touch?fbclid=IwAR1tJZ5Kta42bsJ_OhQArpvCh5R8E9P6RmhOMCg2hZp-a2svqXWdjfj0mLc

el-quique, 2020-04-17:
No implement M600

cosminroger, 2020-04-17:
Well, from what I see that is for the tft35 model and I need to add functionality to tft28/32 one. The tft35 has a different STM32 chip. Thank you for the information tho.

cosminroger, 2020-04-17:
Thank you! I see that you can compile projects there and obtain some bin files so I will take a look!

el-quique, 2020-04-17:
Yes. Using the platform is very easy. The strange thing is that when compiling the project the USB does not work, but I will continue to see how to solve it. Very good luck

el-quique, 2020-04-17:
this link work on Artillery TFT 28 clon MKS:
https://lookaside.fbsbx.com/file/Sidewinder%20X1%20-%20BLTouch.rarx?token=AWwCTtN9oQnY-6bomLV6uvyFvT_SFD-ol4QQRXXIlAdR1iKt-eqQ1FSc3P3FZHKcFbXrDOeYCq2U4hcfSpAfjICkapo3GMb0hb4zwLc2nz7qfi_DFC0K_eHWSyyVy2BnCE898Z2jTqfwCbAGV9ydW0E8

el-quique, 2020-04-19:
> Thank you! I see that you can compile projects there and obtain some bin files so I will take a look!

good news. BigTreeTech now supports TFT MKS:
[Link](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware)

aero4, 2020-04-25:
> 
> 
> > Thank you! I see that you can compile projects there and obtain some bin files so I will take a look!
> 
> good news. BigTreeTech now supports TFT MKS:
> [Link](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware)

How did you manage to upload the firmware?
I copied the MKS folder and MKS firmware from "BIGTREETECH-TouchScreenFirmware/Copy to SD Card root directory to update/" to my SD card ant nothing happened after the boot cycle... when I rename the "MKS_32_V1_4.26.bin" to "MKSTFT28.bin" (same name convention as here) then it updates but never boots...

el-quique, 2020-04-26:
In SD folder MKS with bmp and fonts. And mkstft28.bin

el-quique, 2020-04-26:
I invite you to try my version for Artillery X1.  It is in Spanish by default, but it will be changed from the screen:

https://github.com/el-quique/TFT_Artillery_X1

aero4, 2020-04-26:
> 
> 
> In SD folder MKS with bmp and fonts. And mkstft28.bin

I tried that with the BIGTREETECH FW, but it did not boot...

Ok I will try your version, thanks!

aero4, 2020-04-26:
> 
> 
> I invite you to try my version for Artillery X1. It is in Spanish by default, but it will be changed from the screen:
> 
> https://github.com/el-quique/TFT_Artillery_X1

I tried your version. I copied the folder "MKS", "mkstft28.bin" and "reset.txt" from your repository "TFT_Artillery_X1/Copiar al SD/Theme Dark Clean Edition/" to my sd card and booted the MKS. It is stuck at 100% and rebootng ends that the display is only showng "booting"...
I used a 16GB WIFI-SD card and a 128MB SD card and the problem happens with both...
Any idea? 
I have a MKS TFT32_L V3.0 screen.

aero4, 2020-04-26:
The BIGTREETECH FW only works with HW version 1.4 and I have version 3.0. Therefore it is not working...

el-quique, 2020-04-26:
unfortunately I can only test it with an MKS TFT 28-32 v1.0. back to original TFT MKS FW

el-quique, 2020-04-26:
maybe disconnect wifi modul to upgrade TFT FW

aero4, 2020-04-26:
I don't even have a wifi module. The firmware have to be adapted to V3...

karabas2011, 2020-04-30:
does it support  mks-tft28 v1.2?
I have not bootloader for it and wont risk to blind try

aero4, 2020-04-30:
I don't know, but the bootloader can not be corrupted by this. If the new firmware is not working, just put your old, original mkstft28.bin on the sd card and reboot and everything should be as before....

el-quique, 2020-04-30:
> does it support mks-tft28 v1.2?
> I have not bootloader for it and wont risk to blind try

Hi. Use a Clon MKS TFT v1.0. And BTT use a bootloader for v1.4 and working in my TFT Artillery v1.0