makerbase-mks/MKS-TFT #396: MKS TFT 28 V4.0 with BigTreeTech SKR 1.3 mainboard

KixMan28, 2020-04-17:
My MKS TFT28 V4.0 is not lit up by the SKR board. If I power up the TFT board externally with 12V it shows the nozzle and bed temperatures, I can set their temperatures and I can drive the motors. What should I change in the Marlin 2.0.5 firmware to be able to use the MKS TFT28 without external power?

What I have already did: 
- I connected the MKS TFT28 to the SKR board via the AUX1 connector.
- I have used these settings in Marlin:

`
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
#define CR10_STOCKDISPLAY

`

`
#define` SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define CR10_STOCKDISPLAY

`

What other setting should I try?

KixMan28, 2020-04-20:
My bad, I forgott to move the jumper to the 5V position, as I was using the TFT before with external power adapter. Now the TFT works fine.

closed 2020-04-20 by KixMan28

Bastii92, 2020-07-03:
Why did you set

#define CR10_STOCKDISPLAY

?

Do you use the other display too?