makerbase-mks/MKS-TFT #397: Support for M600 command

Telekmac, 2020-04-17:
M600 is a useful command for filament change and vital if you have filament runout detection. The tft32 doesn't have any support for it and I'd like to see that implemented into the next build as it seems to be an issue multiple people have including myself.

fireflylr, 2020-04-24:
If you simply support the M600 directive, you can do this by configuring the custom button directive in the configuration file.If not, can you come up with a more detailed requirement?What do you want to achieve, etc
--mks

Telekmac, 2020-04-24:
I may try that. But I do think that a big feature you could add is enabling the tft to recieve and respond to firmware messages that require user interaction. One such example being m600. I realize that it runs as a host which is different than a screen running straight off of the mainboard but I think that being able to use commands that require a user interact with with the machine should be supported.

leksus8, 2020-05-05:
Here are some links to using the M600 code in popular firmware Smoothiware and Marlin. 
[http://smoothieware.org/player#stopping-smoothie](url)
[https://marlinfw.org/docs/gcode/M600.html](url)
There is a pause button on the screen during operation. Can you attach the M600 command to this button?

Telekmac, 2020-05-05:
My issue isn't triggering an M600 as my filament run-out sensor and Marlin will do that already. But after m600 is triggered it usually requires user interaction on the screen to reload filament and resume printing. I don't think this could be done by customizing buttons.

el-quique, 2020-05-16:
> My issue isn't triggering an M600 as my filament run-out sensor and Marlin will do that already. But after m600 is triggered it usually requires user interaction on the screen to reload filament and resume printing. I don't think this could be done by customizing buttons.

Hi. See this video for change color. The video is in Spanish:
https://www.youtube.com/watch?v=9zDwaZU5-Eo
Use next code insert in the Layer:xx
G91
G0 X10 Y10 Z20
M0 S120
G0 X-10 Y-10 Z-20
G90
--------------------------------------
Explication
M0 S120 -> 120 second for change filament manually

Telekmac, 2020-05-16:
Well I guess I'll have to put my highschool Spanish classes to use. However I still think that the feature I noted above should be implemented in the next fw version. makerbase said it would be implemented in the next patch multiple versions ago.