makerbase-mks/MKS-TFT #398: TFT35 compiled but black screen

karabas2011, 2020-04-29:
I am able to compile tft35 project ( under IAR 8.30 )
Display is booting as usual but black screen after.
my tft35 is marked as V1.0 version hardware
Need I edit something in code? I am not familiar with IAR
I prefer PlatformIO

Log file in attach
[BuildLog.log](https://github.com/makerbase-mks/MKS-TFT/files/4551938/BuildLog.log)

karabas2011, 2020-04-30:
mks_config.txt does NOT  rename to CONFIG.CUR

el-quique, 2020-04-30:
Hi. Link to IAR software. Thx

karabas2011, 2020-04-30:
> Hi. Link to IAR software. Thx
https://www.iar.com/iar-embedded-workbench/#!?architecture=Arm

karabas2011, 2020-05-09:
firmware does not work
I tried IAR versions 8.30, 7.60, 7.10
What I missed?

el-quique, 2020-05-12:
hi. any tutorial to compile with IAR. Thx

karabas2011, 2020-05-12:
Open Workspace: Go to EWARM folder in mkstft35 folder and open workspace
You will see code tree.
Now make /clean/rebuild 
The result is in EWARM/mkstft35/Exe/ folder

el-quique, 2020-05-13:
In MKS TFT 28/32 the code is different. Thank you

karabas2011, 2020-05-13:
Got it working. You need these steps https://github.com/makerbase-mks/MKS-Robin-Nano-Firmware/wiki/How-to-build-update-nano-firmware

onichan88, 2020-05-17:
It worked, but with which version of IAR?

karabas2011, 2020-05-18:
7.1, 7.5 works

closed 2020-07-07 by karabas2011

venkatkamesh, 2020-08-12:
@karabas2011  can you share the iar 7.1 or 7.5 link ?
really appreciated for your help

Thank you