makerbase-mks/MKS-TFT #403: TFT35 V1.0 Not Connecting To Motherboard

antonrr90, 2020-05-16:
PROBLEM: You can play around with the touch screen menus, but it has absolutely no effect on the Motherboard/Printer it is connected too. Can't Print, Can't Home, Can't Preheat.

SETUP:
   Motherboard : BigTreeTech SKR V1.3
   Connecting to MKS TFT35 V1.0 through the AUX-1 Port.
   MKS TFT35 has been updated to Version1.0.5 Firmware with configuration change to : (cfg_baud_rate:3) to match the 115200 baud rate of my Motherboard.

Q: What Marlin settings are to be used for this device? Which I have yet to find anywhere. 
   #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
is what is currently defined.

Q: Is there any other Marlin 2.0 configurations that may need to be changed to accommodate how the TFT35 interacts with the motherboard?

Q: Why are BTT and MKS both having the same issues with there own TFT35's, and why are they still being sold in this condition. 

You have 10 days to debug, before I send it back as a defect. I don't have time to be your test subject.

antonrr90, 2020-05-16:
Update: Also Tried The BTT SKR V1.3 TFT Pins, No Effect There Either.

fnhhnxfn, 2020-05-19:
check your serial port in marlin and comment out define serial port 2 -1

antonrr90, 2020-05-19:
The board was bad. I had it scoped, and the AUX-1 and TFT had no connection at all, and the EXP1&2 were intermittent. I returned it yesterday. Looking into a more reliable board.

antonrr90, 2020-05-19:
As Far As I Know The TFT35 V1.0 Is Fine, SO IGNORE THIS ISSUE.

closed 2020-05-19 by antonrr90