makerbase-mks/MKS-TFT #405: PB0 as filament runout sensor?

onichan88, 2020-05-16:
Hello, The TFT is preconfigured to use PB0 for power off recovery with UPS, and PB1 with filament runout sensor.

I have a IDEX 3d printer (with 2 extruders and 2 noozles, and my question is that if it can configure the PB0 as a PB1 for the second extruder, as I won't use the power off recovery with UPS?

Regards

fnhhnxfn, 2020-05-19:
hi if you want to use two filament sensor simultaneously you can buy trianglelab filament runout sensor

makerbase-mks, 2020-05-19:
Hi! I suggest you still use the recommended pin.

onichan88, 2020-05-19:
I want to use two filament runout sensor, one per extruder. So if one is getting without pla, it should cold that extruder after regret the pla. If the second is getting without pla, it should cold the second extruder.

So, to differentiate both filament runout sensors, it must have two different pinouts.

This is why i want to use pb0 as a second filament runout sensor.

makerbase-mks, 2020-06-06:
Perhaps the original program does not allow this operation, but you can modify the firmware yourself, we have open sourced the code.

closed 2021-02-24 by mks-viva