makerbase-mks/MKS-TFT #406: TFT35 frozen on babystepping screen

avegao, 2020-05-20:
Hello,

When I’m printing and enter to babystepping screen, the screen freeze and stop the print.

I test this problem in v1.0.5 and the recently launched v1.0.6

McPaul0, 2020-05-30:
I've just update FW on MKS TFT24 from 3.0.3 to 3.0.6 to have the babystepping. and It is the same, it stops every time I press Babystep button.
During heating the babystepping dont work, after that print dont start.

McPaul0, 2020-08-19:
Hey MKS, When you fix this problem?

selvakumaran2016, 2020-10-20:
@mks-viva @makerbase-mks We are waiting this major issue on Babystepping. Screen is feezing when we use babystepping. May I know when you are going to update this.

This issue is for all TFT i hope. I tried in TFT 2.4, TFT 3.5". Both are having same issues

McPaul0, 2020-10-21:
2. repetier firmware , with RepRap firmware on SKR1.4 babystepping works well

selvakumaran2016, 2020-10-21:
With Marlin 2.0 firmware and with MKS TFT this baby-stepping is not working

appliedmike, 2020-10-21:
works with 2.0.7.1

Magiquall, 2021-03-06:
Hello,

Does this work with 2.0.7.2 or the latest Bugfix 2.0.x?
If so, you choose what in "configuration.h" as the parameter for the TFT, because "#define TFT_480x320_SPI" no longer exists in the last Bugfix2.0.x
  #define SHORT_BUILD_VERSION "bugfix-2.0.x"
  #define STRING_DISTRIBUTION_DATE "2021-03-04"
  #define MARLIN_HEX_VERSION 020008
  
 Thank you

pleo993, 2021-03-14:
> Hello,
>
> Does this work with 2.0.7.2 or the latest Bugfix 2.0.x?
> If so, you choose what in "configuration.h" as the parameter for the TFT, because "#define TFT_480x320_SPI" no longer exists in the [...]

Hello i try Bugfix 2.0.x and 2.0.7.2 and still problem... 
(MKS GEN L 2.1 and MKS TFT 35 ..

Magiquall, 2021-03-14:
I let go, I change to BTT, like a lot of people on the forums or Youtubeur, There are some nice things at MKS, but nothing is finalized, its bug everywhere. Makerbase image editor, MKS Tools, well simpa, but it crashes or doesn't work.
I sent Makerbase messages to support and no response.
Luckily I got a response for the screen on Github, if not nothing at all. BTT may not be better in the end, but we'll see. I would try to tweak everything on the table quietly.