makerbase-mks/MKS-TFT #407: TFT70 fw update freezes at 100%

JaydenHLee, 2020-05-21:
I tried to build the source code with IAR 7.5 (and 8.5 too)
And I used 16GB SD card with FAT32. (also tried 32GB SD)
But the screen freezes at "Updating 100%", and it also freezes at "Booting.." after rebooting.

It works well if I use downloadable bin file.
How can I solve this issue?

makerbase-mks, 2020-05-27:
Hi! I think this is caused by the wrong start address. You should set the program start address in the IAR project, because 08000000-0800C000 is the bootloader, so you need to set 0800C000 as the program start address.

JaydenHLee, 2020-05-27:
@makerbase-mks Thank you for your advice.
I could find the address in the source code and tried to change, but it didn't work.
Could you tell me where should I change?

makerbase-mks, 2020-05-29:
Right click after selecting the project  Options ->Linker ->Edit.

makerbase-mks, 2020-06-09:
@JaydenHLee Is it normal?

JaydenHLee, 2020-06-09:
Edit button didn't open any editor.
So I tried to find the file and edit it manually.
But it didn't work and the result was the same.

makerbase-mks, 2020-06-09:
You select project with MKS TFT70 in Left column，Right click and after selecting the project Options ->Linker ->Edit.

closed 2021-02-24 by mks-viva