makerbase-mks/MKS-TFT #415: Technical issue TFT 3.5

nfsaisi, 2020-06-05:
Hi, i have both the 3.5 and the 7.0 , the other day by mistake a power line on a new 3d printer project touched the circuit board of the TFT3.5 and right the way the ARM chip got blown , i'm an electronics service tech so i replaced the ARM processor and all other components checked to be good , 
the only thing because its an empty processor it will not load the Firmware , i need the boot loader , i did notice there is a 6 pin JTAG Header on the Board so just need the info how to load the BOOTLOADER 
Thank you

makerbase-mks, 2020-06-06:
Hi! Your facebook finds us directly, we help you deal with it.

nfsaisi, 2020-06-08:
Thank You, i managed to get the boot loader from your office at Aliexpress , All Good Now !!!

closed 2020-06-08 by nfsaisi