makerbase-mks/MKS-TFT #416: Babystep not working with Vanilla Smoothieware

Mils24, 2020-06-07:
Babystepping does not work with vanilla Smoothieware firmware.

Can you please release source files for the MKSTFT firmware so that the gcode commands linked to the babystep buttons can be changed to suit smoothieware.

Smoothieware g-code for baby stepping is G43.2 Z(babystep height) eg. G43.2 Z0.1 = babystep down Z 0.1mm

makerbase-mks, 2020-06-08:
MKS has open source code, you can refer to 
MKS TFT24：https://github.com/makerbase-mks/MKS-TFT24-Firmware
MKS TFT28-32：https://github.com/makerbase-mks/MKS-TFT28-32-Firmware
MKS TFT35：https://github.com/makerbase-mks/MKS-TFT35-Firmware
MKS TFT70：https://github.com/makerbase-mks/MKS-TFT70-Firmware

Mils24, 2020-06-08:
thank you for quick reply

I have found where the problem is:

#define	BABY_STEP_X_COMMAN(buf, x)		sprintf(buf, "M290 X%.2f\n", x)
#define	BABY_STEP_Y_COMMAN(buf, y)		sprintf(buf, "M290 Y%.2f\n", y)
#define	BABY_STEP_Z_COMMAN(buf, z)		sprintf(buf, "M290 Z%.2f\n", z)

This needs to be:

#define	BABY_STEP_X_COMMAN(buf, x)		sprintf(buf, "G43.2 X%.2f\n", x)
#define	BABY_STEP_Y_COMMAN(buf, y)		sprintf(buf, G43.2Y%.2f\n", y)
#define	BABY_STEP_Z_COMMAN(buf, z)		sprintf(buf, "G43.2 Z%.2f\n", z)

How do I compile the firmware to create the mkstft28.bin file?

Mils24, 2020-06-08:
I have also tried to do this as seen in another thread:

If the screen is printing, you can set it in the configuration file like this:
#how many "More" button display on screen interface

morefunc_cnt: 1
#command setting on button1 ~ 6

morefunc1_cmd: G43.2 Z0.1
morefunc2_cmd: G43.2 Z-0.1
morefunc3_cmd: G43.2 Z0.05
morefunc4_cmd: G43.2 Z-0.05

however when printing the buttons do not appear in the 'more' section.

I have tried this with v.3.05 + v3.06 and both do not work. 

Please tell me how to compile the firmware so I can create my own mkstft_28.bin with the correct gcode values.