makerbase-mks/MKS-TFT #417: MKS-TFT35 Laser interface

Dikto, 2020-06-10:
What about the "Laser" interface for the MKS-TFT35 ?

the "_MKS_TFT35_laser_V1.0.2 Release file.zip_" file seems to contain the interface files for 3D printing.

By comparing with the content of the "_MKS_TFT24_laser_V1.0.2 Release file.zip_" and "_MKS_TFT32_laser_V1.0.2 Release file.zip_" files, only the "_MKS_TFT35_laser_V1.0.2 Release file.zip_" is false.

Is the firmware at least the right one?

fireflylr, 2020-06-11:
Can you describe it more clearly? What is wrong with your tft35 laser engraving firmware?
--mks

Dikto, 2020-06-11:
The problem is not the firmware (well ... I hope), it's the interface.
The one provided in the package is the one for 3D printing.

So I started to make an icon set for CNC / Laser based on the one for the tft32 Laser

But...

From what I could see, the image sizes are all identical for tft24 and tft32 (78x104 px).
For the tft35, there are at least 6 different sizes...

I got the UI template for 3D printing but it does not exist for the CNC/Laser interface and i can't find any example of the CNC or Laser interface for the tft35 to help me.

So I would like to know if MakerBase will eventually release a correct package for CNC/Laser and/or the template for customization...

fireflylr, 2020-06-12:
You can customize the picture you want to modify according to the name of the tft32 laser engraved picture. The picture style of the tft35 button is the same as the tft35 3d printing firmware, so the individual picture size naming is the same as the 3d version.
--mks

Dikto, 2020-06-12:
Thank you for the answer

closed 2021-02-24 by mks-viva