makerbase-mks/MKS-TFT #431: GRBL CTRL+X command in tft35

halny123, 2020-07-16:
I am trying to configure the CTRL+X (0x18: Soft-Reset) command on my screen using the firmware for laser
Tell me how to set up an additional button for sending a command CTRL + X ?
My screen MKS TFT35 using the firmware for laser + board MKS DLC using firmware grbl 1.1
After clicking on the limit switches, you need to reset the grbl using the commands CTRL + X
 
For additional button $H(Run homing cycle) and $X(Kill alarm lock) edited the config file mks_config.txt
But I don't know what to send for the third button  CTRL + X, send code  0x18   no work
>moreitem_button3_cmd:0x18;

--------- mks_config.txt------------------
#set number of  "More" button 
>moreitem_pic_cnt:3

#edit command for 1~4 "More" button,each command must be separated by semicolon ";"
>moreitem_button1_cmd:$X;
>moreitem_button2_cmd:$H;
>moreitem_button3_cmd:0x18;
 
#how many "More" button display on screen interface
>morefunc_cnt:3

#command setting on button1~7
>morefunc1_cmd:$X;
>morefunc2_cmd:$H;
>morefunc2_cmd:0x18;
 ----------------------------

Thanks

fireflylr, 2020-07-25:
You can try the following settings：
moreitem_button3_cmd:ctrl-x;

--mks

m-sja, 2021-02-25:
With the current Firmware it is not possible, cause the TFT "disconnects" if GRBL is in "Alarm Lock" state.
All commands you send via touching any buttons are buffered in the TFT and not send to GRBL.
Recovering from "Alarm-State" is not possible right now with the provided firmware. :-(

@mks please fix this issue

Drayson80, 2021-02-26:
I would also highly support this request!!!
@mks please fix this issue!!!

Option: configure GRBL to "not start in alarm lock" (but from my point of view thats a safety critical modification and I have not tested it up to now)
Nevertheless: @mks please fix this issue!!!

m-sja, 2021-02-26:
@Drayson80 yes and no.
This ist not even a workaround, because if you jog accidentally in the wrong direction  or to far (out of machine space), softlimits will force GRBL in alarm-state. This will also happen while encountering any errors in Check-Mode.
Right now you are forced to hard-reset GRBL witch will also reset the known position as well as volatile stored work- and tool-offsets.

In other words, if this happens during use, your project is screwed :-(

We need an option to bypass the "disconnected" state and force the TFT to send commands without messing around with command-flow-control. Maybe disabling flow-control for user-buttons would be the best and most easy option.

Drayson80, 2021-02-26:
@m-sja, 
thanks for clarification - I understood that this is only on startup the main blocker to get into operation. 
In this case, my suggested workaround is useless - indeed.

@mks, please URGENTLY fix this issue or pls. provide the source so that it can be fixed by the community.

jph290, 2021-03-05:
@MKS-Sean, @makerbase-mks  Will this source be released like the other TFT source?  I am a developer and will be using this for a CNC/Laser and would like to get this working nice.  I am more than happy to modify the source and even commit my changes for others to use.  Are the major differences just the built-in commands?

jvaupoti, 2021-03-07:
@MKS-Sean, @makerbase-mks- i also brought this screen as a CNC control from your aliexpress store, I have the same problem 
NO Z control!!!  Are You doing anything about????

davids5, 2021-03-19:
> @MKS-Sean, @makerbase-mks Will this source be released like the other TFT source? I am a developer and will be using this for a CNC/Laser and would like to get this working nice. I am more than [...]

@MKS-Sean, @makerbase-mks - Same here! Please post the source so we can help make it work correctly with GRBL

void269, 2021-03-29:
Any update on this?

void269, 2021-05-07:
Does anyone know if an earlier version will work?  Really starting to regret buying this thing...since it doesn't work and all...

Drayson80, 2021-05-12:
did this already - now it got a new place in the basement... crap... complete crap... just a few bucks, but regret it...